XCSoar/XCSoar #1159: Kobo Site Files download:  SSL certificate problem

ubx, 2023-03-22:
XCSoar versions having and not having the problem
-------------------------------------------------
NOK: 7.29-Kobe (f2d0e21a95bba191f6e211456b6e6b06331622b2)

OK: 7.29-Kobe (60888ff3109cc1c69fab70359e8a3789c8ab7f96)
OK: 7.28-Kobo

Expected behavior
-----------------
Download requested file (waypoints, airspace, terrain etc.) from server

Actual behavior
---------------
Pop up box:

```
SSL certificate problem: unable to get local
issuer certificate: SSL peer certificate or SSH
remote key was not OK
```

Steps to reproduce the behavior
-------------------------------

1. Connect to Wifi
2. Start XCSoar in fly or sim mode
3. Config → System → Site Files → Map database → Download -> Select `ALPS.xcm` → Download

Do you have any idea what may have caused this?
-----------------------------------------------

Something after the commit 60888ff3109cc1c69fab70359e8a3789c8ab7f96

Do you have an idea how to solve the issue?
-------------------------------------------

Not yet

lordfolken, 2023-03-22:
https://curl.se/docs/sslcerts.html

ubx, 2023-03-22:
Just to carify: if I revert 75efbbef49490b083539a57fae410324d51c3852 the problem is gone.

ubx, 2023-03-22:
Meinsch das?
```
/mnt/onboard # wget https://curl.se/ca/cacert-2022-04-26.pem
Connecting to curl.se (151.101.129.91:443)
saving to 'cacert-2022-04-26.pem'
cacert-2022-04-26.pe 100% |**********************************************************************************************************************************
'cacert-2022-04-26.pem' saved

```
wget --ca-certificate=cacert-2022-04-26.pem https://download.xcsoar.org
Doesn't work on the Kobo:

```
/mnt/onboard # wget
BusyBox v1.31.1.kobo (2020-04-22 11:41:43 EDT) multi-call binary.

Usage: wget [-c|--continue] [--spider] [-q|--quiet] [-O|--output-document FILE]
        [-o|--output-file FILE] [--header 'header: value'] [-Y|--proxy on/off]
        [-P DIR] [-S|--server-response] [-U|--user-agent AGENT] [-T SEC] URL...

Retrieve files via HTTP or FTP

        --spider        Only check URL existence: $? is 0 if exists
        -c              Continue retrieval of aborted transfer
        -q              Quiet
        -P DIR          Save to DIR (default .)
        -S              Show server response
        -T SEC          Network read timeout is SEC seconds
        -O FILE         Save to FILE ('-' for stdout)
        -o FILE         Log messages to FILE
        -U STR          Use STR for User-Agent header
        -Y on/off       Use proxy

```

ubx, 2023-03-22:
The only file I can download is `ogn-device-database.fln`

lordfolken, 2023-03-22:
So i removed the breaking change, as we want an always useable master.
However the upgrade of curl is necessary. But we need to investigate why curl 8 breaks the SSL validation in XCSoar.

ubx, 2023-03-23:
Retested with Libra 2 on master and confirmed that the problem has been solved.

GeekGoneOld, 2023-03-23:
> [17dbbfe](https://github.com/XCSoar/XCSoar/commit/17dbbfed17212682ca40cc0a530b177ceffe2e5e) ca implementation was here.

> But we need to investigate why curl 8 breaks the SSL validation in XCSoar.

I would suggest it is unrelated to the ca implementation commit, which just provides the ca and we know the ca still works.  As @lordfolken says, it is more likely in the SSL validation that uses the ca.

lordfolken, 2023-03-23:
Well the wget connection above proved that the ca is in the correct place, and found by the system. The question is what has changed in curl8 so that it no longer works from xcsoar.

ubx, 2023-03-23:
I have tested it using curl8 on Linux and Android, and it appears to be functioning properly on those platforms. However, it seems that only Kobo is experiencing this issue

MaxKellermann, 2023-05-03:
9fee4b4d0af079e8be11632b2d10040d26250bf3

closed 2023-05-03 by MaxKellermann