XCSoar/XCSoar #1162: F-droid builds fail to install

lordfolken, 2023-03-26:
[image]

Fdroid builds fail to install.

@linsui do you have an idea what's wrong there? 

Releases from download.xcsoar.org install fine on the same devices.

ManDay, 2023-03-26:
The device is a Pixel 3a with LineageOs 19.1 `sargo:12/SP2A.220505.008`. I attach the logcat upon attempting to install the package (which shows above error)
[log.txt](https://github.com/XCSoar/XCSoar/files/11071691/log.txt)

linsui, 2023-03-26:
I don't know. 7.27 works but 7.28 not. They are built in the same way. Is there anything changed?

lordfolken, 2023-03-26:
https://github.com/XCSoar/XCSoar/blame/v7.28/android/testing/AndroidManifest.xml 
the androidsdk was raised, the document provider was added and some details. i guess one of those is breaking the build.

lordfolken, 2023-03-26:
Is there an easy way to reproduce the build environment, so we could build offline and test?

linsui, 2023-03-26:
The easiest way is using the gitlab ci. [This](https://monitor.f-droid.org/builds/log/org.xcsoar/149）is the log. Is there anything weird?

lordfolken, 2023-03-26:
Logcat from install
```
3-26 21:13:30.118  1428  1531 E PackageManager: ERROR: could not load root hash from incremental install
03-26 21:13:30.118  1428  1531 E PackageManager: Required verifier is null
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: Exception reading file:///data/app/vmdl456817158.tmp
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: com.android.server.pm.PackageManagerException: Targeting R+ (version 30 and above) requires the resources.arsc of installed APKs to be stored uncompressed and aligned on a 4-byte boundary
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: 	at com.android.server.pm.parsing.PackageParser2.parsePackage(PackageParser2.java:172)
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: 	at com.android.server.pm.parsing.PackageParser2.parsePackage(PackageParser2.java:150)
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: 	at com.android.server.integrity.AppIntegrityManagerServiceImpl.getPackageArchiveInfo(AppIntegrityManagerServiceImpl.java:624)
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: 	at com.android.server.integrity.AppIntegrityManagerServiceImpl.handleIntegrityVerification(AppIntegrityManagerServiceImpl.java:300)
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: 	at com.android.server.integrity.AppIntegrityManagerServiceImpl.-$$Nest$mhandleIntegrityVerification(Unknown Source:0)
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: 	at com.android.server.integrity.AppIntegrityManagerServiceImpl$1.lambda$onReceive$0(AppIntegrityManagerServiceImpl.java:183)
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: 	at com.android.server.integrity.AppIntegrityManagerServiceImpl$1.$r8$lambda$D-6BanMaetK6OzqqHwb9xKEKfbs(Unknown Source:0)
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: 	at com.android.server.integrity.AppIntegrityManagerServiceImpl$1$$ExternalSyntheticLambda0.run(Unknown Source:4)
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: 	at android.os.Handler.handleCallback(Handler.java:942)
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: 	at android.os.Handler.dispatchMessage(Handler.java:99)
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: 	at android.os.Looper.loopOnce(Looper.java:201)
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: 	at android.os.Looper.loop(Looper.java:288)
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: 	at android.os.HandlerThread.run(HandlerThread.java:67)
03-26 21:13:30.124  1428  1630 W AppIntegrityManagerServiceImpl: Cannot parse package org.xcsoar
03-26 21:13:30.124  1428  1531 I PackageManager: Integrity check passed for file:///data/app/vmdl456817158.tmp
03-26 21:13:30.311  1428  1531 W PackageManager: Failed parse during installPackageLI: Targeting R+ (version 30 and above) requires the resources.arsc of installed APKs to be stored uncompressed and aligned on a 4-byte boundary
03-26 21:13:30.313  1428  1531 D PackageInstallerSession: Marking session 456817158 as failed: -124: Failed parse during installPackageLI: Targeting R+ (version 30 and above) requires the resources.arsc of installed APKs to be stored uncompressed and aligned on a 4-byte boundary
03-26 21:13:30.314   973  1582 E installd: Couldn't opendir /data/app/vmdl456817158.tmp: No such file or directory
03-26 21:13:30.314   973  1582 E installd: Failed to delete /data/app/vmdl456817158.tmp: No such file or directory
```

linsui, 2023-03-27:
Looks like we didn't align the apk. We should take the output/ANDROID/opt/build/aligned.apk instead of the output/ANDROID/opt/build/unsigned.apk. I guess the bumped SDK makes the unaligned apk not work anymore.

linsui, 2023-03-27:
Fixed in https://gitlab.com/fdroid/fdroiddata/-/merge_requests/12815. You can test the [apk](https://gitlab.com/linsui/fdroiddata/-/jobs/4007872038/artifacts/file/tmp/org.xcsoar_149.apk).

ManDay, 2023-03-27:
Installing from the APK doesn't work for me, it says "App not installed as package appears to be invalid." - didn't see anything helpful in logcat at a glance.

lordfolken, 2023-03-27:
So its a two step process
First sign the package, then zipalign it.
https://stackoverflow.com/questions/22058210/why-unaligned

linsui, 2023-03-27:
It's not signed. You need to sign it first.

lordfolken, 2023-03-27:
Yeah but our ```build/android.mk```  line 320 forward, first takes the unaligned package , aligns and then signs it. Therefore destroying the alignment as per above article.

https://stackoverflow.com/questions/22058210/why-unaligned

https://stackoverflow.com/questions/22058210/why-unaligned

linsui, 2023-03-27:
It sounds weird to me. All apks on F-Droid are aligned before signed.

lordfolken, 2023-03-27:
Since we are using apksinger the current implementation is correct.

lordfolken, 2023-05-05:
It installed on a brand new pixel. 7.29 is missing however.

linsui, 2023-05-05:
It will be available in few days.

MaxKellermann, 2023-05-25:
So... is this really a XCSoar bug? Our APKs are aligned and signed properly, aren't they?

linsui, 2023-05-25:
It's not a XCSoar bug and it's fixed already.

closed 2023-05-25 by lordfolken