XCSoar/XCSoar #1173: Build fails for PI with ‘FT_Error_String’ was not declared in this scope

ubx, 2023-04-11:
XCSoar versions having and not having the problem
-------------------------------------------------
XCSoar version are you using when seeing the problem:
Current head (5be2f8d5d6a655733db202cc8e06623d91c727c0)

Expected behavior
-----------------
compiling on the Raspberry Pi 4:
`make`

Make a build for Pi

Actual behavior
---------------
compiling on the Raspberry Pi 4:
`make`

```
  CXX     output/UNIX/dbg/src/ui/canvas/freetype/Init.o
src/ui/canvas/freetype/Init.cpp: In function ‘FT_FaceRec_* FreeType::Load(const char*)’:
src/ui/canvas/freetype/Init.cpp:48:36: error: ‘FT_Error_String’ was not declared in this scope
   48 |                              path, FT_Error_String(error));
```
and

```
  CXX     output/UNIX/dbg/src/ui/canvas/freetype/Init.o
src/ui/canvas/freetype/Init.cpp: In function ‘FT_FaceRec_* FreeType::Load(const char*)’:
src/ui/canvas/freetype/Init.cpp:49:36: error: ‘FT_Error_String’ was not declared in this scope
   49 |  
```  

Steps to reproduce the behavior
-------------------------------
compiling on the Raspberry Pi 4:
`make`

Do you have any idea what may have caused this?
-----------------------------------------------
Probably commit 12129c921c9768933c7cca839ecbeff5106eabcf

Do you have an idea how to solve the issue?
-------------------------------------------
Wait for a PR

ubx, 2023-04-12:
The issue was resolved after upgrading from FreeType 2.9.1 to the latest version, 2.13.0.

closed 2023-04-12 by ubx

MaxKellermann, 2023-04-12:
This is not resolved. The wrong version should have been detected and rejected.