XCSoar/XCSoar #1177: HGT Map support

XCNav, 2023-04-15:
XCSoar version
--------------

All Versions

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

HGT Map support

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

LXNAV Devices are using high definition HGT Maps. You can download them on LXNAV webpage. Even if LXNAV would make you pay for it, this would be a great alternative to the current open source maps.

MaxKellermann, 2023-04-15:
You forgot to post the link to the file format specification, and you did not demonstrate the superiority of this file format and the data. You need to be very convincing for people to spend their free time on helping a commercial company earn money.

lordfolken, 2023-04-15:
Our srtm maps are higher resolution than theirs. What's the point?

lordfolken, 2023-04-15:
Maybe we should do another release so people see the new contrast settings. IMO map renderer is a lot more capable than is visible with current settings.

lordfolken, 2023-04-16:
@XCNav what are you missing with current maps?

XCNav, 2023-04-16:
> @XCNav what are you missing with current maps?

[image]

I am happy with the default XCSOAR maps. This would just be an alternative for another map source, as the .hgt maps are free to download ont the lx website 
The picture shows about the same area in the ALPS_highres.xcm and the LX HGT Map taken from a screenshot of the LX Website. It seems likebthere is a slightly higher resolution on LXSite and more details.

lordfolken, 2023-04-16:
That screenshot looks strangley low res:

Here is: 
XCSoar (current master)
ALPS_highres.xcm
Terrain: Vibrant

[image]

lordfolken, 2023-04-16:
I saw that now almost all of western europe and usa has dem1 data available. Current is dem3. 
http://viewfinderpanoramas.org/Coverage%20map%20viewfinderpanoramas_org1.htm

lordfolken, 2023-04-16:
https://github.com/XCSoar/mapgen/issues/59

MaxKellermann, 2023-04-17:
You still didn't reply to my request for a link to the file format specification, and it looks like you won't ever. Without this, your feature request will never be implemented.

closed 2023-04-17 by MaxKellermann