XCSoar/XCSoar #1178: Support for Bluetooth LE engine sensor box

fastdown101, 2023-04-16:
XCSoar version 7.9
--------------

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
I would like to have support for a engine sensor box in the Android implementation. I'm the developer of the sensor box.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
The sensor box is aimed to be used by powered paragliding pilots. The sensor box gathers engine related sensor values as well as atmospheric pressure, which should be shown in xcs. Relying on the engine while flying powered paragliders, I find it crucial to have engine "health" measurements like Exhaust Gas Temperature (EGT) eadings. With xcs, the sensor box would be a full blown flight computer.

What have I done so far?
-------------------------------------------------------------------
I branched locally from master. Added my "driver" implementation down to "void DeviceDescriptor::OnEngineSensboxSensors(...)" in AndroidSensors.cpp. I guess this is where the the parsed sensor values reach the native c++ code? How to proceed from there?

Where do I need help?
-------------------------------------------------------------------
How can I add support to get the sensor values displayed? Some further calculations of the sensor values are needed too, as well as calculations depending on some user configuration. How can I upstream my branch? Any help will be greatly appreciated!

Some details on the sensor values
-------------------------------------------------------------------
The sensor box does not do any calculations. For instance, the engine Revolutions Per Minute (RPM) value has to be calculated by xcs from revolutions per second. Depending on the used engine there might be 2 ignitions per revolution. So the user has to configure xcs correctly depending on the engine used, to get correct readings for the RPM value.
It would be great to have the following values displayed:
engine Revolutions Per Minute (RPM), two high temperature sensors up to 1024 Degree Celsius e.g. Cylinder Head Temperature (CHT)  and Exhaust Gas Temperature (EGT), QNH based on the barometric pressure, ambient temperature and optional sensor health flags

MaxKellermann, 2023-04-16:
How is this data going to be transmitted? As BLE characteristics?

First you need to add missing methods for transmitting those values to Java `class SensorListener`, and implementations to forward them into native C++ code to `class NativeSensorListener`, and the C++ side needs to have the same methods in `class SensorListener`.

As you already found out, the C++ `class DeviceDescriptor` implements that interface, and needs a way to store those values; right now, all other sensor values are stored in `struct NMEAInfo` somewhere. This struct name is a historic name for "stuff we parsed from NMEA", even though this is not NMEA. In this struct, maybe add a new sub-struct like `struct EngineState`, and put your values there. Be sure to implement methods `Reset()`, `Complement()` and `Expire()` properly.

Once this infrastructure is in place, add support for your sensor box to Java `class BluetoothSensor`. This is where BLE characteristics are parsed and passed to `class SensorListener`.

This was 99% of the work. The rest, displaying it, is easy: add some InfoBoxContent code so you can show it in an InfoBox. Later, you might want to record the data to show a graph... but let's talk about that when everything else is done.

(Maybe the above instructions should be added to the developer manual? PR welcome.)

fastdown101, 2023-04-16:
Hello Max,

thanks for your quick reply.

Data is in one BLE service with one characteristic, right. I implemented all the java stuff (UUID, parsing `BluetoothGattCharacteristic` etc.) you mentioned, down to the `DeviceDescriptor`. It compiles without errors. Going to tackle the NMEAInfo now, maybe without a substruct first.

I can not push my branch, how is the procedure to push? I have a public ssh key uploaded to my github account... I think it would be more easy talking details, with code to point at.

Thanks and best regards,

Daniel

MaxKellermann, 2023-04-16:
You need to create a fork of XCSoar in your GitHub account (the "Fork" button at the top right), and you can push anything there, because it's your personal derivative of XCSoar; from there, you can create a pull request to this official repository. Since you already have some code, I agree that would be the right way forward.

fastdown101, 2023-04-16:
Great news :)
I got it up and running.
For testing, I used the `Outside Air Temperature ` to display the temperature of the barometer from my sensor board, works like a charm. Even the configured timeout when sensor data does not come in periodically is working. That makes me think, that I figured out how to use the NMEAInfo struct.

So, now how to make a new custom InfoBox? Where to look for simple examples? 

It seems like once a BLE device is disconnected (hard by pull the power plug), there is no new scan and thus no reconnect?

My code is Forked in -> [klick](https://github.com/fastdown101/XCSoar/tree/AddEngineSensorBox)

XCNav, 2023-04-16:
This is a very good Idea and would also be usable in powered gliders or with fes sustainers. Why not adding an optional engine page in Look>Pages for the bottom or main section.

fastdown101, 2023-04-17:
> Why not adding an optional engine page...
I like the idea of adding a page, but up to now I am struggling to get a configurable InfoBox...

Speaking of InfoBoxes, I found a `src/InfoBoxes/Factory.cpp` which has a huge `MetaData` array. Do I have to create a `/src/InfoBoxes/Engine.{cpp,hpp}` based on one of the other files there and then add a entry in the `meta_data` array to get a InfoBox for one of my sensor values?

MaxKellermann, 2023-04-17:
Yes, add an entry to `Factory.cpp` referring to a callback function in `Engine.cpp`.

fastdown101, 2023-04-17:
I'm confused, the xcs developer manual says: If I am a InfoBox, I should get data from the `InterfaceBlackboard` with a `BlackboardListener`. But most of the other examples in `InfoBoxes` use `const NMEAInfo &basic = CommonInterface::Basic();` to access data. I assume the `CommonInterface::Basic()` method is okay? If it is okay, at which rate/interval are the InfoBoxes updated?

MaxKellermann, 2023-04-17:
Read what `CommonInterface::Basic()` does!

MaxKellermann, 2023-04-17:
> at which rate/interval are the InfoBoxes updated?

After each CalculationThread run, which gets invoked each time a new GPS location is received. That trigger is implemented in MergeThread.cpp
https://github.com/XCSoar/XCSoar/blob/a0b0ad682209f25be076b7e33686e4cb3df5f6e5/src/MergeThread.cpp#L101-L102

fastdown101, 2023-04-17:
> Read what CommonInterface::Basic() does!

Okay, I see :)

> After each CalculationThread run, which gets invoked each time a new GPS location is received.

Hm, there is no shortcut to update or some kind of mechanism to update, when new sensor values are present? I understand that there is probably expensive calculations running a task/glide-computer, but a trigger for simple sensor values would be great.

Thanks for your patience, I'm probably asking stupid simple questions all over again...

MaxKellermann, 2023-04-17:
Your questions are not stupid, because the XCSoar code base is, like all code bases, under-documented, and if knowledge is only in my head, there needs to be a way to extract it, and you're doing that.

Much of what you see is 20 year old XCSoar code which I overhauled 10-or-so years ago. Many names are old and deprecated, but the code behind it was somewhat modernized. Prior to my overhaul, everything was more confusing than now; but more importantly, my overhaul has made it much more reliable. It's still confusing and needs more work, but life is so short...

There is a shortcut: if GPS is unavailable, non-GPS sensors must still be updated, and for that reason, `TriggerVarioUpdate()` exists:
https://github.com/XCSoar/XCSoar/blob/a0b0ad682209f25be076b7e33686e4cb3df5f6e5/src/Protection.cpp#L44-L49
Since this can be called from arbitrary threads, `MainWindow::SendGPSUpdate()` first uses a `UI::Notify` instance to translate this into a callback in the main (=UI) thread, which then calls `MainWindow::OnGpsNotify()`; that calls `UIReceiveSensorData()`.
https://github.com/XCSoar/XCSoar/blob/a0b0ad682209f25be076b7e33686e4cb3df5f6e5/src/UIReceiveBlackboard.cpp#L18-L35
Here, the blackboard is copied from the (thread-safe but slow) DeviceBlackboard to the (non-thread-safe only valid in UI thread) InterfaceBlackboard. After that, it does some this and that, and finally there's a check whether to update InfoBoxes under certain conditions (if nobody else does it, i.e. if there's no GPS).

Running the CalculationThread is only useful for GPS data, because without a GPS location, none of these calculations make sense. But even if there is a GPS, it might make sense to update InfoBoxes if no GPS fix has been received for a certain amount of time, let's say 2 seconds, as a safe fallback to ensure that the user never sees old data from non-GPS sensors.

However, that should be an improvement for later; for now, I believe it's fine with the current implementation. Updating engine data more than once a second is not useful, it only drains battery for unnecessary redraws.

fastdown101, 2023-04-17:
Okay, that's great so I pick your brain somewhat more :)

I assumed reading about a Blackboard, that this is a publish subscribe mechanism, but it's not, right? At least not in a sens, that subscribers get notified on new data they subscribed to. 
The stuff you described sounds not that intuitive to me, but I'm also very new to this code base...

> Updating engine data more than once a second is not useful

I fully agree. My sensor box does not send data more often either. But how to be sure, that the GPS is updated with 1 Hz? The android implementation `locationManager.requestLocationUpdates(locationProvider, 1000, 0, this);` says for the given 1000 ms interval, that it is the `minimum time interval between location updates in milliseconds`. I guess that is the best we get for now?

Anyway, I'm super happy with my progress so far. Thanks for your help. I managed to get all my sensor data printed in InfoBoxes.  Now its time for the icing on the cake :) 

I need persistent data for:
- The time the engine is running. It should be tracked for maintenance purposes. Reset of the engine hours should be possible, too.
- The engine RPMs scaling factor. It has to be applied by the user. A simple drop down with 3 choices should do.
- Titles of the engine temperature InfoBoxes. Those should be configurable by the user. A drop down with 3 choices should do. Cylinder Head Temperature (CHT), Exhaust Gas Temperature (EGT) and Water Temperature (WTR), what else would you monitor?!?

Further, the enhancement of the BLE implementation to allow for reconnect of lost sensors, would be great. If I remember correctly, the android implementation is capable of scanning and communicating in parallel. Usually connection to my sensor board is pretty stable, but it would be nice to have automatic reconnect of known sensors...

With best regards, 
viele Grüße,

Daniel

MaxKellermann, 2023-04-17:
> I assumed reading about a Blackboard, that this is a publish subscribe mechanism, but it's not, right?

Somewhat yes, somewhat no. There are several copies of the blackboard, one lockless one for the CalcThread, one lockless one for the UI thread (InterfaceBlackboard), and DeviceBlackboard is thread-safe (and has a per-device copy which gets merged into one by MergeThread after each sensor change).
(If you have a handset with software renderer i.e. without OpenGL, then the map renderer has its own blackboard (MapWindowBlackboard) because it runs inside the DrawThread and thus needs its own lockless copy.)
XCSoar has so many blackboard copies because it's important to have safe lockless access to all data.

Device drivers write to DeviceBlackboard, and the UI obtains values from InterfaceBlackboard. You rarely need to know more than that, unless you dig deep into the XCSoar core.

InfoBoxes get invoked on each GPS update, and the map renderer also gets updated automatically.

Only if you implement a modal dialog you need to take care for registering a listener; the LiveBlackboard class (implemented by InterfaceBlackboard) has methods to register such a listener so you get called back whenever something changes.

XCNav, 2023-04-17:
> * Titles of the engine temperature InfoBoxes. Those should be configurable by the user. A drop down with 3 choices should do. Cylinder Head Temperature (CHT), Exhaust Gas Temperature (EGT) and Water Temperature (WTR), what else would you monitor?!?
>

fastdown101, 2023-04-18:
> Only if you implement a modal dialog

But something like a `InfoBoxPanel` is not a modal dialog, right?

> What about Fuel level?

Most of the time a paramotor is not stable enough in flight for reliable fuel level metering. I mean, you ware the motor as a backpack...fuel in the tank is all over the place. One could think about a flow sensor to get amount of burned fuel, but those sensors are not very accurate and small in diameter, so in my opinion not suitable for the use case paramotor. I stick to the good old mirror in my hand to look at the tank...

> electric PPG (e.g.open PPG) think about Watts ,Amps

On those machines, reading of Watts, Amps, Voltage etc. is usually already monitored and made available to the user. I'm talking about small 2-stroke engines, air or water cooled...

MaxKellermann, 2023-04-18:
> But something like a `InfoBoxPanel` is not a modal dialog, right?

Yes, this is a modal dialog. You can use `BlackboardListener` here, and several of those panel implementations do.

Scumi, 2023-04-18:
> * what else would you monitor?!?

Oil temperature and pressure are also standard measurements for an aircraft engine. I would definitely monitor those as full throttle take-offs at low oil temperature can easily kill the engine.
What about manifold pressure of constant speed prop implementations?

MaxKellermann, 2023-04-18:
When you design a protocol, be sure to document what shall be transmitted when a certain value is/becomes invalid. For example, if you define a payload that contains several values, allow a new product which does not implement all attributes because it doesn't have all sensors to use the same protocol.
But before you design a new protocol, check whether such a protocol already exists. There are numerous standardized BLE protocols.

MaxKellermann, 2023-04-18:
`uint16` - what is the byte order? How do you specify missing values? What is then unit for the temperature? It's Kelvin, isn it? Which is why you said it's unsigned, right?
`float32` - what is the exact bit format of this floating point value, and which byte order? What happens with infinities, NaN etc.?

It's not as simple as it sounds. Every corner case must be well-defined.

fastdown101, 2023-04-18:
Okay, I see. There is lots more to do before I issue a pull request onto XCSoar master...

MaxKellermann, 2023-04-18:
You can avoid lots of awkward complexities by never using floating point. Only use fixed-point, and wrap everything in integers. Choose one by order and keep using it everywhere. BLE usually uses little-endian. When you write code, be sure it works on both little-endian and big-endian CPUs.

Also it may be helpful to avoid signed integers. Temperatur - that one is easy, juse use Kelvin! But what kind of value is really stored? Kelvins? Deci-Kelvins? Centi-Kelvins? If you store Kelvins, be aware that you will never be able to be more precise than 1K. How much precision will you need? Which range will you need?
There are existing BLE protocols for temperature sensors - check what they do.

Most importantly - really, I can't emphasize it enough - how do you encode missing/invalid values? Will you use a "magic" special value? Are you sure it doesn't interfere with a reasonable value? Or will you use a "flags" field that specified which fields are present in the rest of the payload?

For magic values, a counter-example: some people like to encode "locationn unavailable" as "0°N 0°E", but that is a real location. Using that "magic value" will make you unable to fly over that location. You probably never will anyway, but in any case, using that is an extremely bad engineering decision.

Another example: use -1 is "altitude not available". But what about Dutch or Isreali pilots who live are below sea level? Bad design!

Which "reveolutions per second" value means "sensor not present"? Don't use 0 - because 0 then can either mean "not turning right now" or "sensor reading not available". Don't be ambiguous! The difference is important!

fastdown101, 2023-04-20:
I'm struggling to get a drop down list with text in it for the RPM scaling factor.
After selecting a text in the drop down, I would like to act on the selected text, e.g. scale the RPM by a factor of 2 or so. How to do this? What I have so far:
https://github.com/fastdown101/XCSoar/blob/AddEngineSensorBox/src/InfoBoxes/Content/Factory.cpp
```
 {
    N_("Engine Revolutions Per Minute"),
    N_("RPM"),
    N_("Engine Revolutions Per Minute."),    
    IBFHelper<InfoBoxContentRevolutionsPerMinute>::Create,
  },
```

https://github.com/fastdown101/XCSoar/blob/AddEngineSensorBox/src/InfoBoxes/Content/RevolutionsPerMinute.cpp

```
static constexpr InfoBoxPanel rpm_scale_panel[] = {
  { N_("Scale"), LoadRevolutionsPerMinuteScalePanel },  
  { nullptr, nullptr }
};

const InfoBoxPanel *
InfoBoxContentRevolutionsPerMinute::GetDialogContent() noexcept
{
  return rpm_scale_panel;
}
```

With this code I get a `LoadRevolutionsPerMinuteScalePanel` widget when I press the RPM InfoBox. But there I'm stuck.

https://github.com/fastdown101/XCSoar/blob/AddEngineSensorBox/src/InfoBoxes/Panel/RevolutionsPerMinuteScaling.cpp

```
class RPMOffsetButtons final : public OffsetButtonsWidget {
public:
  using OffsetButtonsWidget::OffsetButtonsWidget;

protected:
  // virtual methods from OffsetButtonsWidget 
  void OnOffset(double offset) noexcept override;
};

void
RPMOffsetButtons::OnOffset(double offset) noexcept
{
  ActionInterface::OffsetManualMacCready(Units::ToSysVSpeed(offset));
}

std::unique_ptr<Widget>
LoadRevolutionsPerMinuteScalePanel([[maybe_unused]] unsigned id)
{
  const auto step = GetUserVerticalSpeedStep();
  return std::make_unique<RPMOffsetButtons>(UIGlobals::GetDialogLook().button,
                                                  GetUserVerticalSpeedFormat(false, true),
                                                  step, 5 * step);
}
```

Obviously the buttons are no drop down list^^
How to get a custom drop down list to chose custom values from?

MaxKellermann, 2023-04-20:
RPM scaling factor? Huh? What do you really want to do here?

> Obviously the buttons are no drop down list^^

If you want to have a drop down list, then why did you choose to use an OffsetButtonsWidget? I mean, that code didn't write itself by accident. This was your explicit decision, but why?

MaxKellermann, 2023-04-20:
> I just copied and altered the "MacCready settings" stuff without understanding much...

Don't do that. If you copy code, at least copy code that does the thing you want to do. You copied something that doesn't do what you want to do, and surprisingly, it didn't do what you wanted to do. Well, duh!

If you want to display a drop down list (formerly called "combo box"), you need an "edit" control with a DataField that fits your use case. To display such a control in a Widget, use RowFormWidget. You'll find plenty of examples of RowFormWidget in the code base.

Do you really want to attach that to the InfoBox? Those panels are meant to be a short-cut to features that exist somewhere else.

Let's look at this another way - before you allow the user to edit something, that "something" must exist. Where does this feature exist? Rephrased: what value will be displayed initially, and what happens if the user modifies it? Don't say: then XCSoar will scale, that's too inaccurate. What part of XCSoar will scale? The InfoBox? How does the InfoBox know how to scale? Is scaling inside the InfoBox really a good idea? What if some other part of XCSoar accesses the value, will it see an unscaled value? Will the user have to enter this value again each time he restarts XCSoar? What if the user has two RPM sensors, will those always have the same scaling?

This stuff is difficult, but there is no shortcut. You need to write code that has answers to all of these questions (and many more questions), and all of these answers must be by design, not by accident.

MaxKellermann, 2023-04-20:
> This is in my mind a no brainer as a calculation, even for a InfoBox... I have seen worse in the code...

Then your code has no chance to ever get merged into XCSoar. If you don't want that, fine, but please say so early on.

fastdown101, 2023-04-20:
Have you had a look into my code? Lets discuss on code base what to do, to get merged.
So the requirement is: User hast to chose to scale RPM by a factor of 2.
Where to implement? How to implement? Drop down list, just a button...?

MaxKellermann, 2023-04-20:
> Have you had a look into my code? Lets discuss on code base what to do, to get merged.

No, why would I? You didn't submit it for review. And before I spend more time on explaining things to you, I want to be confident that you'll finish this properly, and not do bad short cuts. I wasted so much time on helping people who then gave up before any usable code came from them, and your previous post sounded like a red flag.

fastdown101, 2023-04-20:
> I wasted so much time on helping people who then gave up before any usable code came from them

Okay I get that. 

I'm trying to understand the code base and add support for my sensor box. I'm totally willing to do some extra hops through loops, to be as generic as possible. This might be beneficial for other people, adding their engine sensor box, without to much hassle with the software. 

You think I need something like the `AtmosphericPressure` or `Temperature` class for RPM? Because I'm using those for my pressure and temperature readings.

MaxKellermann, 2023-04-20:
> Where to implement? How to implement? Drop down list, just a button...?

This should probably a per-device setting; an unscaled (incorrect) value must never be stored in `NMEAInfo`, because that will someday lead to somebody using that wrong value. Make the code robust against such mistakes!

Per-device settings are in `struct DeviceConfig` (maybe add an `unsigned rpm_scale` there?), and therefore it should be edited in `DeviceEditWidget`.
There's `Profile/DeviceConfig.cpp` which is responsible for loading/saving `DeviceConfig` values from/to the profile.

MaxKellermann, 2023-04-20:
About
> Drop down list, just a button...?

That doesn't matter so much. It is important to have the (invisible) code model properly designed, that can't be changed easily, but a UI can be changed easily at any time.
But a button doesn't seem to make much sense. Clicking a button should cause some action to occur, but you want the user to see and set a value. A button is the wrong tool for the job.
Use `RowFormWidget::AddInteger()` to create an edit control.

closed 2023-12-01 by MaxKellermann