XCSoar/XCSoar #1183: COM Ports not showing up in versions 7.24+ (Windows, possibly UNIX as well)

Witloofboer, 2023-04-17:
XCSoar versions having and not having the problem
-------------------------------------------------

Versions of XCSoar up to version 7.23 show a list of COM ports. From 7.24 on list does not exist, no serial device can thus be coupled. When the COM-port has been used in a previous version, the port appears in the newer versions as well (tested Windows only).

Bug is present in some UNIX versions as well.

Expected behavior
-----------------

A list of COM-ports to select from.

[image]

Actual behavior
---------------

No options to select a serial port.

[image]

Steps to reproduce the behavior
-------------------------------

Open 7.24+ version in Windows.

Do you have any idea what may have caused this?
-----------------------------------------------

Might be related to: https://github.com/XCSoar/XCSoar/issues/888

From release notes 7.24:

[image]

Does not explain behavior in UNIX though.

Do you have an idea how to solve the issue?
-------------------------------------------

No(t yet)

MaxKellermann, 2023-05-07:
In 7.23, COM ports 1..10 were always displayed, no matter if they exist or not.
In 7.24, a check was added to show only COM ports that actually exist.

Does your complaint mean that you do have a computer with 10 serial ports? Or did you never wonder why XCSoar shows 10 of them? You did not describe which ones you think were really missing.

Witloofboer, 2023-05-08:
> In 7.23, COM ports 1..10 were always displayed, no matter if they exist or not. In 7.24, a check was added to show only COM ports that actually exist.
>

I haven't checked the latest release, but this was not the behavior that I encountered. I had two COM-ports connected to a Windows machine, both FTDI-based USB to serial converters. None of them showed up in XCSoar, but they were accessible using TeraTerm/Python/...

I thought it might have had something to do with my configuration/PC as it happened in both my Windows and Linux environments, so I've checked with someone else on a different PC, and they experienced the same behavior.

> Does your complaint mean that you do have a computer with 10 serial ports?

No, the complaint means that there was, for me at least, no other way to connect a new serial device than reverting back to an older version, using the drop down list to add it and then moving back to the newest XCSoar version.

> Or did you never wonder why XCSoar shows 10 of them?

No, I didn't, 10 always seemed plenty to me as well.

> You did not describe which ones you think were really missing.

The USB/UART & USB/RS232 interfaces I was using at the time, that were definitely working with other software.

I'll download the latest release and check again, as I've seen that some changes have been made. Thanks in any case!