XCSoar/XCSoar #1186: BLE write failure on version 7.7 and later

ryanwoodie, 2023-04-19:
XCSoar versions having and not having the problem
-------------------------------------------------
Android 7.7 and later exhibit the problem

Works on 7.6

Expected behavior
-----------------
Connect to BLE Port (HM10) device, using lxnav protocol, and nmea sentences should be TX by XCsoar over BLE, to be received by BLE device.

Eg:
$PLXV0,CONNECTION,W,VSEVEN*52
$PLXV0,NMEARATE,W,2,5,1,60,30,0,0*6F

Actual behavior
---------------
The first byte (ie. 24 aka "$") is sent, after that there is a "port write" failure for the next chunk of bytes. Extremely rarely, maybe 1 in 100 presses of the MC up/down, a sentence might come through.

Steps to reproduce the behavior
-------------------------------
Connect to ESP32 BLE device with HM10 UUID in XCsoar, using LXNAV protocol.
Monitor output on BLE device.
Enable debug logging on device in device setup if desired.
Press MC up/down.

Do you have any idea what may have caused this?
-----------------------------------------------
There were a number of changes from 7.6 to 7.7 in BluetoothGattClientPort file that might have caused this: https://github.com/XCSoar/XCSoar/compare/v7.6...v7.7

Not sure if this happens on other BLE devices, or only ESP32 ones. Have replicated the problem with multiple ESP32 boards, and multiple Android phones.

Do you have an idea how to solve the issue?
-------------------------------------------
Revert the offending change(s) maybe in 7.7 in BluetoothGattClientPort (which no longer exists as it has been moved to file(s) starting with HM10).

If I get things setup to compile code, I might be able to figure it out myself, but the vastly more knowledge devs here might know at quick glance what the problem is!

ryanwoodie, 2023-04-20:
I have been able to replicate the problem apart from ESP32, using nRF connect app as a BLE server on one phone, and XCsoar on the other. When debug mode is on the device, however, using this method the problem goes away and it will write quite reliably.

closed 2023-05-03 by MaxKellermann

ryanwoodie, 2023-05-04:
Thanks @MaxKellermann for looking at this. The port_write error message is now gone, but it is still not sending data successfully over BLE. Instead of sending just an "$" now, now all that comes through upon hitting reconnect is a "D". I did get one stray "$PLXVC,INFO,R*0" chunk come through just once however in about 20 reconnects. Nothing comes through when MC buttons are clicked.

MaxKellermann, 2023-05-04:
Write a new bug report, describe exactly what you see - "D", what "D" ..?