XCSoar/XCSoar #1188: Flight date

PiRomanet, 2023-04-21:
Hello,
Since I use revision 7.28, the flights files are recorded with a wrond date.
For example, a flight made on april 19 2023, is recorded september 03 2003.
Not only the file name is wrong, The date inside the file is wrong as well.

 At the same time, files are recorded in a different directory, which is maybe normal even if it makes use a bit more difficult. The main concern is really the date which makes the files unusable for OLC or Netcoupe.
 
 Hope it is easy to understand and to fix.

MaxKellermann, 2023-04-21:
The good old GPS rollover bug. This is a bug in your GPS, not in XCSoar.

delta217, 2024-04-05:
Hello, I re-open this discussion, my flight of yesterday was registered as 2004-08-19.
And this is not a GPS problem, I use my phone for backcountry skiing, and dates are correct  !
Thanks.

PiRomanet, 2024-04-05:
BonjourLe fait est que nombre d'application utilisent la date donnée par le reseau téléphone, qui est toujours correcte.XCsoar utilise la date donnée par le GPS du téléphone. Il se trouve que sur certains modèles, dont le S9 par exemple, il y a un beug, qui fausse la date. (Problème de dépassement de  valeur).Je crains qu'il ne faille pas attendre de correction de la part de xcsoar, et pas non plus de la part du fournisseur du téléphone. Pour ma part, j'ai du changer de modèle pour continuer à enregistrer les vols.Malgré tout, sur mon S9, la date a de nouveau été correctement gérée après quelques mois de défauts. On peut le vérifier avec une application comme "GPS data" par exemple.Bon vol malgré toutPierreEnvoyé depuis mon appareil Galaxy
-------- Message d'origine --------De : Jean-Philippe ***@***.***> Date : 05/04/2024  10:52  (GMT+01:00) À : XCSoar/XCSoar ***@***.***> Cc : PiRomanet ***@***.***>, Author ***@***.***> Objet : Re: [XCSoar/XCSoar] Flight date (Issue #1188) 
Hello, I re-open this discussion, my flight of yesterday was registered as 2004-08-19.
And this is not a GPS problem, I use my phone for backcountry skiing, and dates are correct  !
Thanks.

delta217, 2024-04-05:
Bonjour,
Intéressant... par contre GPS data donne l'heure, mais pas la date.
J'ai le problème avec un S10 qui a remplacé mon S7.
Donc, bidouillage des fichiers IGC pour pouvoir les enregistrer sur planeur.net and co.
Merci pour ces explications.

lordfolken, 2024-04-05:
> Hello, I re-open this discussion, my flight of yesterday was registered as 2004-08-19. And this is not a GPS problem, I use my phone for backcountry skiing, and dates are correct ! Thanks.

This is absolutely the GPS Rollover bug.
https://en.wikipedia.org/wiki/GPS_week_number_rollover

Check out "GPS Test" app from Google Play to see what date/time your device reads from the GPS system.

PiRomanet, 2024-04-05:
Sur mon S9, j'ai la date en bas de la page Info.C'est en la regardant ce matin que j'ai vu que la date était redevenue correcte.Quand j'ai eu le même problème,  il y avait une vingtaine d'année d'ecart, comme pour toi aujourd'hui. PierrrEnvoyé depuis mon appareil Galaxy
-------- Message d'origine --------De : Jean-Philippe ***@***.***> Date : 05/04/2024  12:15  (GMT+01:00) À : XCSoar/XCSoar ***@***.***> Cc : PiRomanet ***@***.***>, Author ***@***.***> Objet : Re: [XCSoar/XCSoar] Flight date (Issue #1188) 
Bonjour,
Intéressant... par contre GPS data donne l'heure, mais pas la date.
J'ai le problème avec un S10 qui a remplacé mon S7.
Donc, bidouillage des fichiers IGC pour pouvoir les enregistrer sur planeur.net and co.
Merci pour ces explications.

delta217, 2024-04-05:
> This is absolutely the GPS Rollover bug. https://en.wikipedia.org/wiki/GPS_week_number_rollover
> 
> Check out "GPS Test" app from Google Play to see what date/time your device reads from the GPS system.

Hello, I checked the date : 05-04-24 (in ddmmyy)
I changed the format to to yymmdd to see any effect...?
My S10 was introduced on the market in April 2019 !

lordfolken, 2024-04-05:
Samsung released a phone with an old firmware in the gps chip. That is why. There where plenty of updates due to this from them. 
Potentially you aren't running the latest version https://www.sammobile.com/firmwares/archive/ has firmwares for your phone. 
In any case, there is nothing that xcsoar can or will do. It will believe the date your gps chip reports.

ubx, 2024-04-09:
> This is obviously a GPS bug, but XCSoar (which is definitely great, thank you to his developers) has no choice than to manage it !

IMHO, XCSoar is not responsible for bugs in the underlying platform, such as the buggy S10. I would not invest any time to fix the GPS rollover bug in XCSoar, even if it would only require a single line of code.

Instead, I would suggest asking Samsung for a new firmware.

delta217, 2024-04-09:
Well, if you read Wikipedia on this famous GPS bug (https://en.wikipedia.org/wiki/GPS_week_number_rollover#2137_occurrence), you see that plenty of systems were affected, including for commercial flights. For sure, they had no choice than to fix the issue in their system. Also, on different forums, I read that a pilot mentioned that his S7 was affected, while mine and my former S5 were not !

To answer your suggestion, yes, I asked Samsung on this question... Not sure to have an answer and a solution...?

I do like XCsoar, and promote it with my students. I would be really disappointed to switch to another system because of this problem.

lordfolken, 2024-04-10:
Please use another piece of Software as xcsoar doesn't seem to meet your needs.

Potential candidates:
http://lk8000.it/
https://akaflieg-freiburg.github.io/enroute/
https://apps.apple.com/us/app/iglide-lite/id400664090
https://naviter.com/seeyou-navigator/
https://www.foreflight.com/
https://magazine.weglide.org/weglide-copilot-real-time-data-on-your-phone/

delta217, 2024-04-10:
Hello,
Thanks for the list. I am setting seeyou-navigator. But before giving up XCSoar (I really like it !), I will invert peripherals A and B. A is internal sensors of the phone, and B is the bluetooth connected to the flarm. It might help ?

lordfolken, 2024-04-11:
I'd go as far as disabling the internal sensors, and using the Flarm exclusively. XCSoar can fallback to the other sources, potentially trashing your log file. The internal sensors don't really give you anything that the flarm doesn't have.

delta217, 2024-04-11:
The phone has a pressure sensor, and I would like to keep it.

lordfolken, 2024-04-11:
So has your Flarm.

F-Phi, 2024-04-15:
I would rather keep my internal pressure sensor as well, as this has some advantages over the flarm system, i.e. more temperature resitent

delta217, 2024-04-22:
Moving FLARM to peripheral A solved the problem.

JBKingdon, 2025-07-18:
Just adding my voice to the list of parties interested in a solution to this problem. I'm just getting back into soaring after a few years and have run straight into the gps week rollover issue. I'm using a bluetooth gps module and accept that it would be ideal to have it output the correct data. Unfortunately I don't think that is an option at this point. I looked into replacements, but bluetooth gps's seem to have become rather more limited in selection and certainly a lot more expensive. If a software work-around was possible that would be very much appreciated.