XCSoar/XCSoar #1193: Long InfoBox values displayed too small

Scumi, 2023-04-25:
XCSoar versions having and not having the problem
-------------------------------------------------
Issue starts at 7bf77a3b380689f64f0ddfb4ea2dbace072f3115

Expected behavior
-----------------
Long infobox values taking all available space and are therefor shown with a normal size.

[image]

Actual behavior
---------------
Infobox values have a margin around them, which reduces the size of long values.

[image]

Steps to reproduce the behavior
-------------------------------
Use infoboxes that can contain longer values (i.e. frequency infoboxes).

Do you have any idea what may have caused this?
-----------------------------------------------
7bf77a3b380689f64f0ddfb4ea2dbace072f3115

Do you have an idea how to solve the issue?
-------------------------------------------
Revert commit (quick fix) or improve logic.

Scumi, 2023-04-25:
@lordfolken can you have a look?

lordfolken, 2023-04-25:
Is this infobox frequency only?

Scumi, 2023-04-25:
According to the linked PR, there could be other infoboxes as well. I need to check which infoboxes might handle "long" values.

lordfolken, 2023-04-25:
I mean the calculation always was for "1234m" so not really long values. However this is unchanged... sinc ever.

Scumi, 2023-04-25:
Okay, understood. Nevertheless, the issue starts with this commit for me.
I will check other infoboxes (if I find similar ones) and test a few commits "around" that one if that helps.

lordfolken, 2023-04-25:
Whats strange here: It should overflow and not remain small.

lordfolken, 2023-04-25:
Ok, so this is by design:

https://github.com/XCSoar/XCSoar/blob/b1309a96f7bd6707dc369f6264ffb5e8d5872b73/src/InfoBoxes/InfoBoxWindow.cpp#L108-L112

Should the value not fit in the width, the above code will select the small value (the one for the comments size font.

lordfolken, 2023-04-25:
@Scumi does the above PR fix the issue for you?

Scumi, 2023-04-26:
Yes, it does.

closed 2023-04-28 by MaxKellermann