XCSoar/XCSoar #1202: No data read-out from ATR833

Scumi, 2023-05-02:
XCSoar versions having and not having the problem
-------------------------------------------------
Issue starts at 43c24921f1fb0537c35b81983aeb72946e3d05d4

Expected behavior
-----------------
XCSoar can read-out data from ATR833 radio device and display frequencies in respective infoboxes.

Actual behavior
---------------
No data read-out.

Steps to reproduce the behavior
-------------------------------
Compile XCSoar version at 43c24921f1fb0537c35b81983aeb72946e3d05d4 and after.

Do you have any idea what may have caused this?
-----------------------------------------------
43c24921f1fb0537c35b81983aeb72946e3d05d4

Do you have an idea how to solve the issue?
-------------------------------------------
Not yet, started troubleshooting.

MaxKellermann, 2023-05-02:
Oh no! Goodness you tested this early before I could release the bug.

Scumi, 2023-05-02:
I was curious about the improvements you did as they look nice and simple.

Further information: Feedback when changing the passive frequency or pressing the exchange frequency button on the ATR833 is transferred immediately to XCSoar. 

My best guess is currently, that there might be an issue with the return cases for ALLDATA, ACK, NAK and ALIVE.

MaxKellermann, 2023-05-02:
I rewrote some rather ugly and fragile code, but apparently I made a mistake. I really need to write an emulator so we can have unit tests for the ATR driver.
Do you have a way to capture communication between XCSoar and the ATR? (with a working version of XCSoar, of course.) Having that would be very helpful for writing the emulator.

Scumi, 2023-05-02:
I have no RS232 sniffer unfortunately. I connect to the device either via XCSoar or CuteCom. Do you have means to capture comms? I could send the radio to you if that helps.

MaxKellermann, 2023-05-02:
Looks like this was a very stupid mistake. Please check if the new commit fixes your problem.

I need to make additional changes to the driver because a lot of bugs remain (not mine - the original submission had those bugs already) - will you be available for more testing?

closed 2023-05-02 by MaxKellermann

Scumi, 2023-05-02:
Just checked, it partially works again.

Before improvements: XCSoar immediatly shows device read-out when started (due to XCSoar sending ALIVE and REQUESTDATA).
Currently: XCSoar shows again data readout, but only after frequency has been changed once on the device or the exchange frequency button has been pressed.

Yes, I am available for more tests. After this driver is done, I will re-wire my setup and also check KRT2 and if there are ways of reading out data from this one as well.

MaxKellermann, 2023-05-02:
Strange, because my fix didn't change anything about the SETACTIVE/SETSTANDBY handlers, and those are responsible for receiving manual changes!

Scumi, 2023-05-02:
As said, it works perfectly once the device has sent anything by itself at least once.
It looks more like `EnableNMEA()` is not triggered at the beginning, but you also did not change anything there as far as I see (but I only looked briefly).

MaxKellermann, 2023-05-02:
Try the following (new) program to capture data:
```
make output/UNIX/bin/SplicePorts
output/UNIX/bin/SplicePorts tcp 4353 /dev/ttyS1 115200 2>&1 |tee /tmp/log
```
Then configure XCSoar to connect the ATR driver to 127.0.0.1 port 4353 (if XCSoar runs on a different device, configure the IP address of the device running SplicePorts) - the second set of parameters describes the port/baud where the real ATR is connected.
This program forwards all data between the incoming TCP connection and the serial port, and dumps it to stderr (and to /tmp/log).

Scumi, 2023-05-02:
Done. What I did:

1. Start XCSoar (first five lines of the log)
No regular transmissions of ALIVE and REQUESTDATA via `OnSysTicker()`
2. Set passive frequency on ATR833
This one starts the data read-out and the regular transmissions, but only for one or two cycles. Then it stops again.
3. Press Exchange Frequency button on ATR833
The data read-out and regular transmissions are started again for one or two  cycles, then it stops again.

[ATR833_troubleshooting.txt](https://github.com/XCSoar/XCSoar/files/11372915/ATR833_troubleshooting.txt)

MaxKellermann, 2023-05-02:
I think I got it now.

Scumi, 2023-05-02:
Confirmed.

Scumi, 2023-05-02:
Everything works flawlessly. Impressive code improvements, at least for me.