XCSoar/XCSoar #1205: Implementation of Fenix Vario

XCNav, 2023-05-03:
XCSoar version
--------------

All

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Add driver for Fenix Vario

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

Right now the Phenix Vario works with the LX Driver, but no Down/Upload of IGC files is possible. As there is already a working driver in LK8000 it should be a minor issue only.
https://github.com/LK8000/LK8000/blob/master/Common/Source/Devices/devRCFenix.h

MaxKellermann, 2023-05-03:
Where's the protocol documentation?

Fenix-C, 2023-05-03:
> Where's the protocol documentation?

Hi Max.
It's all here. 
If I can help with anything, I'm happy to help. 

https://www.rc-electronics.eu/downloads-gliding/manuals-gliding/

https://www.rc-electronics.eu/wp-content/uploads/2022/01/RC_Electronics_protocol_v10.pdf

MaxKellermann, 2023-05-03:
Am I missing something? Where's the part that describes how IGC file transfer works?

MaxKellermann, 2023-05-03:
The LK8000 driver has no code for IGC file transfer. And that's the only thing this feature request is about. I don't get it.

Fenix-C, 2023-05-04:
Hi Max,
Mea culpa. I misunderstood a colleague. A remedy is already being worked
on. It's very important to me.
Thank you for your response.
Leos

lordfolken, 2023-05-04:
The current version of the LXNAV Data Port Specification, seems to have a function to up and download files to their current variometers (s8x,s1xx).
XCSoar has no current implementation for that yet.

Fenix-C, 2023-05-05:
A colleague adds. Mea culpa. I'll let you know.

MaxKellermann, 2023-06-29:
Impossible to implement right now. Reopen when it becomes possible.

closed 2023-06-29 by MaxKellermann