XCSoar/XCSoar #1208: BLE not writing properly (still) on version 7.7 and later

ryanwoodie, 2023-05-04:
XCSoar versions having and not having the problem
-------------------------------------------------

Which XCSoar version are you using when seeing the problem?
Latest Android github action compilation: https://github.com/XCSoar/XCSoar/actions/runs/4873589613

Which XCSoar version is the last one you know that did not show the problem?
7.6

Expected behavior
-----------------

Connect to BLE Port (HM10) device, using lxnav protocol, and nmea sentences should be TX by XCsoar (upon connection or when pressing MC buttons) over BLE, to be received by BLE device.

Eg. upon connection:
$PLXV0,CONNECTION,W,VSEVEN52
$PLXV0,NMEARATE,W,2,5,1,60,30,0,06F

Actual behavior
---------------

With FLARM Protocol this is what is received over BLE (when reconnecting):
23:01:32.660 -> 0
23:01:32.660 -> 
23:01:32.660 -> $PFLAV,R*33

With LXNAV Protocol (when reconnecting)
23:04:12.481 -> D

Just 1 time, this was received with LXNAV protocol (when reconnecting):
22:50:04.746 -> *49
22:50:04.746 -> $PLXVC,INFO,R*0

Steps to reproduce the behavior
-------------------------------

Connect to ESP32 BLE device with HM10 UUID in XCsoar, using LXNAV protocol.
Monitor output on BLE device.
Press MC up/down or press reconnect.

Do you have any idea what may have caused this?
-----------------------------------------------

There were a number of changes from 7.6 to 7.7 in BluetoothGattClientPort file that might have caused this: https://github.com/XCSoar/XCSoar/compare/v7.6...v7.7

Do you have an idea how to solve the issue?
-------------------------------------------

Revert the offending change(s) maybe in 7.7 in BluetoothGattClientPort (which no longer exists as it has been moved to file(s) starting with HM10).

ryanwoodie, 2023-05-04:
Will test with 2 phones (one acting as a the BLE server) later to see if it replicates on that.

MaxKellermann, 2023-05-08:
I bought a generic HM10 and just tried to reproduce your problem. Everything works as expected with 7.29. No data lost, no corrupt data, nothing.

MaxKellermann, 2023-05-08:
@ryanwoodie, several phones have no problem and no data loss with HM-10.
Please try again with XCSoar 7.6 and confirm that 7.6 really works and that 7.29 does not.

ryanwoodie, 2023-05-09:
Hi @MaxKellermann,
Sorry, you are correct that both 7.29 and 7.30 are working correctly with BLE. I had tried many methods of processing the BLE data in my arduino code previously with 7.28, and it was arbitrarily left off using an approach (that had worked with 7.6 and LK8000), that was directly streaming the chunks to a variable vs. buffering till receiving the "\n" and the variable was overwriting itself before being read (on my primary phone, but not a second phone actually which must have had a slightly slower transmit speed).

I sincerely apologize for the false alarm on this, and thank you greatly for fixing the issue. Can I reimburse you for the HM10?

Ryan

P.S. Out of curiosity, the BLE fixes seemed to happen in 7.30, but 7.29 also works. If you happen to know off the top of your head, what was the thing that fixed it in 7.29?

closed 2023-05-09 by ryanwoodie

MaxKellermann, 2023-05-09:
This problem was fixed by commit https://github.com/XCSoar/XCSoar/commit/1ba0c7594dbc54edd9f3001d093121cb4b78f490 as noted in your bug report https://github.com/XCSoar/XCSoar/issues/1186 - the problem was that Android refuses parallel operations on a BLE device, and that commit serializes all operations.