XCSoar/XCSoar #1209: Circle airspace shape not showing correctly based on zoom

kobedegeest, 2023-05-05:
XCSoar versions having and not having the problem
-------------------------------------------------

since 7.29

Expected behavior
-----------------
Airspace look not changing on zoom lvl. And deffenetly not dissapearing.

Actual behavior
---------------
Circle formed airspace changes look depending on size of circle and zoom lvl.
When filled airspace circle changes in square, only fill no border. When zooming in.
When no fill (border or border with padding) airspace just dissapears when zooming in.

Both things shown in pictures below.

Two different circles.
[image]
One cirlce dissapeared.
[image]
Second circle changed in square.
[image]

Not all circles do this at the same zoom so i suppose it is based on radius and zoom

Steps to reproduce the behavior
-------------------------------

Find a circle shaped airspace in pan mode and zoom in on it. Also happens in non pan mode.
If needed i can provide airspace file i used but i think it is not related to the used file.

Do you have any idea what may have caused this?
-----------------------------------------------

Latest update.

Do you have an idea how to solve the issue?
-------------------------------------------

Nope

MaxKellermann, 2023-05-05:
What device, graphics chip and OS version is that?

lordfolken, 2023-05-06:
I guess its not showing, when it doesn't fit on the screen?

kobedegeest, 2023-05-06:
I assume it thinks the vieuw is completly in the circle so it draws no edge and only the fill as square.
But it is not ... can pan around and not find the circle. Also are such optimimazations even needed? And is it something xcsoar does or my device? I assume i can by pas by just changing all circles by polygons or two arcs or something.
Can someone reproduce on different device or not?

kobedegeest, 2023-05-07:
> I guess its not showing, when it doesn't fit on the screen?

or if you mean with does not fit on screen, radius circle just bigger than width of screen that is also not the case i can see the full square so circle would deffenetly fully fit on the screen.

MaxKellermann, 2023-05-08:
I can reproduce a similar rendering problem on a Pixel 2.

closed 2023-05-08 by MaxKellermann