XCSoar/XCSoar #1210: Altitude should be on the right and speed on the left side

lordfolken, 2023-05-06:
XCSoar version
--------------
All

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
XCSoar displays the final glide bar on the left, and speed related information on the right. 

This is contrary to pretty every EFIS / Navigation system in aviation. 

B747:
[image]

lxnav:
[image]

garmin:
[image]

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

gratefulfrog, 2026-08-31:
I confirm that glide bar on right collides with button icons on Kobo.
I suggest that the config for the vario and final glide bars offer the options: Left, Right, None, rather than simply On/Off.
I suggest that it be possible to put both glide bars on the left, or the right, next to each other.

lordfolken, 2026-08-31:
These buttons are not on per default on the kobo. The intended use case for the kobo is the up and down swipe gestures.

gratefulfrog, 2026-08-31:
Thank you for your reply. I personally like the buttons, that's how I discovered that they cover the newly places glide bar.  In any case, I do use the swipe gestures, but can never remember which is plus and which is minus - that's why I like the buttons.