XCSoar/XCSoar #1211: Low res or missing Icon on Desktop versions of XCSoar (WIndows/Linux)

lordfolken, 2023-05-06:
XCSoar version
--------------
7.29 (win64 / linux)

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

When alt-tabbing between applications, usually the application icon is shown.

Linux just shows a generic "application" icon, windows shows one that is of low resolution.
[image]

MaxKellermann, 2023-05-16:
Which Linux build? SDL or X11/EGL?

lordfolken, 2023-05-16:
```
Starting XCSoar XCSoar v7.32-Linux~git#326ecec41d
GLX config: RGB=8/8/8 alpha=8 depth=24 stencil=8
GL vendor: AMD
GL version: 4.6 (Compatibility Profile) Mesa 20.3.5
GL renderer: AMD SIENNA_CICHLID (DRM 3.49.0, 6.1.0-0.deb11.6-amd64, LLVM 11.0.1)
```

MaxKellermann, 2023-05-17:
I still don't know

lordfolken, 2024-06-09:
Two different problems. The xcsoarswift.ico in Data/bitmaps, didn't take modern windows icons into account.

For linux, we are missing a Desktop Entry file that would allow xcsoar be registered in menus of a desktop, such as gnome, both as a menu entry and an application icon.

commit bb361c4f (2024-06-10): Data/bitmaps/xcsoarswift.ico: icon with sizes up to 256px
Fixes #1211
closed 2024-06-10 by lordfolken