XCSoar/XCSoar #1221: waypoints not shown

OBrown92, 2023-05-16:
We got some issues with a waypoint file. Many waypoints are not shown. Can someone debug the following file?
[waypoints.cup](https://www.soaringspot.com/de/download-contest-file/4226-27243) 

The Turnpoints 002-006 for example are missing.

OBrown92, 2023-05-16:
They disappear after a restart of xcsoar. Btw. we are using the latest Android Version from playstore.

MaxKellermann, 2023-05-16:
The invisible waypoints have no elevation, and your map file cannot provide a fallback value, therefore XCSoar discards the waypoints whose elevation cannot be resolved. This has nothing to do with a new "2022" column order.

closed 2023-05-16 by MaxKellermann

lordfolken, 2023-05-16:
I think this is mainly due to the asynchrpnous loading of the terrain. The fallback altitude is calculated before terrain elevation data is available. 
https://github.com/XCSoar/XCSoar/issues/900