XCSoar/XCSoar #1224: mac os compilation failing

qubolino, 2023-05-22:
XCSoar versions having and not having the problem
-------------------------------------------------
7.31

Expected behavior
-----------------
successful compilation

Actual behavior
---------------
```
 make dmg V=2
clang++ src/Dialogs/Weather/NOAAList.cpp -c -o output/OSX64/dbg/src/Dialogs/Weather/NOAAList.o -MD -MF output/OSX64/dbg/src/Dialogs/Weather/NOAAList.d -MT output/OSX64/dbg/src/Dialogs/Weather/NOAAList.o -Og -ffast-math -g   -std=c++20 -fno-threadsafe-statics -fmerge-all-constants -fcoroutines-ts -fvisibility=hidden  -ObjC++ -fobjc-arc -stdlib=libc++ -Wall -Wextra -Wwrite-strings -Wcast-qual -Wpointer-arith -Wsign-compare -Wundef -Wmissing-declarations -Wredundant-decls -Wmissing-noreturn -Wvla -Wno-missing-field-initializers -Wcast-align -Werror -Wno-error=deprecated-declarations   -I./src/unix -I./output/include -isystem ./output/src/boost_1_82_0 -DBOOST_NO_IOSTREAM -DBOOST_MATH_NO_LEXICAL_CAST -DBOOST_UBLAS_NO_STD_CERR -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_SYSTEM_NO_DEPRECATED -DBOOST_NO_STD_LOCALE -DBOOST_LEXICAL_CAST_ASSUME_C_LOCALE -DBOOST_NO_CXX98_FUNCTION_BASE -I./src -I./src/Engine -I./output/OSX64/include -DHAVE_POSIX -DHAVE_VASPRINTF -DEYE_CANDY -DXCSOAR_TESTING -isystem /include -D__ASSERT_MACROS_DEFINE_VERSIONS_WITHOUT_UNDERSCORES=0        -I./src/Terrain/jasper/..          -D_THREAD_SAFE -isystem /usr/local/include -isystem /usr/local/include/SDL2 -DENABLE_SDL    -DUSE_APPKIT  -DUSE_MEMORY_CANVAS          -D_THREAD_SAFE -isystem /usr/local/include -isystem /usr/local/include/SDL2 -DENABLE_SDL     -isystem /usr/local/Cellar/libtiff/4.5.0/include -isystem /usr/local/Cellar/zstd/1.5.5/include -isystem /usr/local/Cellar/jpeg-turbo/2.1.5.1/include -DUSE_LIBTIFF -DUSE_GEOTIFF -isystem /usr/include/geotiff -DENABLE_COREGRAPHICS      -DUSE_APPKIT    -D_THREAD_SAFE -isystem /usr/local/include -isystem /usr/local/include/SDL2 -DENABLE_SDL     -DUSE_MEMORY_CANVAS            -isystem /usr/local/Cellar/fmt/10.0.0/include  -isystem /usr/local/Cellar/c-ares/1.19.0/include        -isystem /usr/local/include/lua   -DZLIB_CONST     -isystem /usr/local/Cellar/libsodium/1.0.18_1/include -isystem /usr/local/Cellar/fmt/10.0.0/include -DBOOST_JSON_STANDALONE -Wno-#warnings         -mmacosx-version-min=12.0 -target x86_64-apple-darwin
fatal error: error in backend: Cannot select: intrinsic %llvm.coro.size
clang: error: clang frontend command failed with exit code 70 (use -v to see invocation)
Apple clang version 13.1.6 (clang-1316.0.21.2.5)
Target: x86_64-apple-darwin
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
clang: note: diagnostic msg:
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang: note: diagnostic msg: /var/folders/7w/mq36knvn6874129l6n8t1rzm0000gn/T/NOAAList-fcd991.mm
clang: note: diagnostic msg: /var/folders/7w/mq36knvn6874129l6n8t1rzm0000gn/T/NOAAList-fcd991.sh
clang: note: diagnostic msg: Crash backtrace is located in
clang: note: diagnostic msg: /Users/qubo/Library/Logs/DiagnosticReports/clang_<YYYY-MM-DD-HHMMSS>_<hostname>.crash
clang: note: diagnostic msg: (choose the .crash file that corresponds to your crash)
clang: note: diagnostic msg:

********************
make: *** [output/OSX64/dbg/src/Dialogs/Weather/NOAAList.o] Error 70
```

Steps to reproduce the behavior
-------------------------------
```
brew install \
  automake autoconf libtool \
  pkg-config \
  quilt \
  librsvg \
  imagemagick gettext sox \
  fmt \
  sdl2 \
  libsodium \
  freetype \
  libpng libjpeg-turbo \
  libtiff libgeotiff proj \
  c-ares \
  curl \
  lua
```
```
make dmg
```

Do you have any idea what may have caused this?
-----------------------------------------------
not really

Do you have an idea how to solve the issue?
-------------------------------------------
sadly not

MaxKellermann, 2023-05-22:
This is not a XCSoar but, but a clang bug. Your toolchain is broken. We can't fix that.

qubolino, 2023-05-23:
fair enough. would you be willing to provide a mac os binary then? or help fix this so that i can build it for the community?

MaxKellermann, 2023-05-23:
No, I don't have a Mac and I'm not interested in "fixing" the Mac build, whatever that means with a known-buggy compiler.

qubolino, 2023-05-23:
i understand and sense your frustration.

in that case i can only suggest you amend https://xcsoar.readthedocs.io/en/latest/build.html by reomving the mac os section to avoid false hopes and time wasted for everyone

lordfolken, 2023-05-23:
Every commit is built also on osx. See for example:
https://github.com/XCSoar/XCSoar/actions/runs/5057935633

In the wokflow file there is a mentioning of your bug:
https://github.com/XCSoar/XCSoar/blob/35f47267854ccfeddeea12ee4d1e38edc2668f43/.github/workflows/build-native.yml#L456-L480