XCSoar/XCSoar #1226: NMEA 'AcftType' field is hexadecimal

lyusupov, 2023-05-23:
XCSoar versions having and not having the problem
-------------------------------------------------

7.31 and earlier versions

Expected behavior
-----------------

<br>

[**FTD-012 DATA PORT INTERFACE CONTROL DOCUMENT (ICD)**](https://flarm.com/wp-content/uploads/man/FTD-012-Data-Port-Interface-Control-Document-ICD.pdf)

<br>
<br>

[image]

Actual behavior
---------------

https://github.com/XCSoar/XCSoar/blob/master/src/Device/Driver/FLARM/StaticParser.cpp#L153

```
  unsigned type = line.Read(0);
  if (type > 15 || type == 14)
    traffic.type = FlarmTraffic::AircraftType::UNKNOWN;
  else
    traffic.type = (FlarmTraffic::AircraftType)type;
```

Do you have any idea what may have caused this?
-----------------------------------------------

No

Do you have an idea how to solve the issue?
-------------------------------------------

No

lordfolken, 2023-05-23:
Its unfortunately completely brain-dead:

From the Configuration document TD-014-FLARM-Config:

ACFT:

Configures the aircraft type used by FLARM. This setting will affect the internal
algorithms, as FLARM’s motion prediction is aircraft type specific

**Warning**: Contrary to the hexadecimal convention in PFLAA, the aircraft type is
given as a decimal value here.

Doh!

lordfolken, 2023-05-23:
So. This is not a problem. The char value read from the pflaa line is assigned to an unsigned type. Both the comparison here:

https://github.com/XCSoar/XCSoar/blob/35f47267854ccfeddeea12ee4d1e38edc2668f43/src/Device/Driver/FLARM/StaticParser.cpp#L153-L157

and the enum retrival here:
https://github.com/XCSoar/XCSoar/blob/35f47267854ccfeddeea12ee4d1e38edc2668f43/src/FLARM/Traffic.hpp#L32-L48

Will accept the hex value and return the match for the corresponding decimal value.

MaxKellermann, 2023-05-23:
This bug report confused me. Was this about a theoretical problem (that turned out to be bogus after all?) or was a real problem observed with a real FLARM?

lordfolken, 2023-05-23:
Well the reporter doesn't say. 
However i did some experiments with https://github.com/ra1fh/airtraffic
if you change the aircraft_type to a char ie "c". 
the plane type in the flarm dialog does not get recognized as a UAV, but is UNKOWN. 

So I'm not sure that the reading of the char actually converts it to a hex value. It probably converts C to the ascii value which is of course higher than 15. 

  unsigned type = line.Read(0);

Therefore the Plane Type in the Traffic details dialog is wrong for values higher than 9. (UAV,AIRSSHIP..etc)

closed 2023-05-25 by MaxKellermann