XCSoar/XCSoar #1235: Lenovo M7 TB-7206F crashes in PAN mode

antonka2000, 2023-05-30:
XCSoar versions having and not having the problem
-------------------------------------------------
I started having the Problem in Season 2022.
actually I am using 7.33

Actual behavior
---------------
When I am in PAN Mode XCSoar crashes after a few moves.
Zooming and moving makes no difference.

Do you have any idea what may have caused this?
-----------------------------------------------
The 2nd Gen Tablet has only 2GB of RAM.
Maybe that is a problem.

MaxKellermann, 2023-05-30:
That's impossible. Version 7.3.0 does not exist.

matejuh, 2023-05-31:
By coincidence one of our members also reported same issue today. Same tablet, latest XCSoar version 7.3.3
He will try older releases and once I will meet him I will try to get logcat report.

matejuh, 2023-05-31:
7.33 I meant 🤦

lordfolken, 2023-05-31:
Please try older releases until you have something stable.
Https://downloads.xcsoar.org/releases

Scumi, 2023-06-17:
[lenovo_crash.txt](https://github.com/XCSoar/XCSoar/files/11778728/lenovo_crash.txt)

Here is a logcat output from the crash. It only crashes if topography or terrain is enabled.

MaxKellermann, 2023-06-17:
@Scumi, your crash is in the graphics driver shipped by Lenovo. It is possible that this is a bug in the graphics driver, but maybe it's a bug in XCSoar which only leads to a crash with this driver. I'm unable to find out from here; I'd need to have a hardware that reproduces the crash.

Scumi, 2023-06-21:
I can make one of ours available in August. I try to check older versions to see, if there was a certain point where the crashes started.
@matejuh Do you have one before that date for Max to investigate?

lordfolken, 2024-02-09:
@Scumi is this still an issue?

antonka2000, 2024-02-20:
sorry to say - yes. And even worse:
I tried to use no map. - No problem and no crash.
Then I tried the high resolution map. - Now it cashes shortly after starting xcsoar.
So I cannot use it at all at the moment with this device.
Version: 7.41

MaxKellermann, 2024-02-20:
Nobody will be working on this until we have such a device.

Scumi, 2024-02-20:
Will send you one after the weekend. Where can I find your address?

lordfolken, 2024-02-20:
@Scumi https://max.kellermann.name/impressum.html

MaxKellermann, 2024-03-04:
Don't send a device to me. I no longer work on XCSoar.

lordfolken, 2024-09-04:
So i bought one of these devices. I can confirm it crashes on zoom / pan function.

lordfolken, 2024-09-04:
the addr2line points to ./src/ui/canvas/opengl/Dynamic.hpp:37

lordfolken, 2024-09-04:
The bug is present in all 7.XX versions down to 7.5. 6.8.17 is working.