XCSoar/XCSoar #1236: XCSoar The version of the hardened plug-in is too low, not support the new Android system!

nrbray, 2023-05-30:
XCSoar v7.31-Android
Same issue, two phones: Android 12 Moto g(50) and Android 10 Hisense A7 CC

Crash with above message on Android 10, Crash with no message on Android 12, a few seconds after the following setup.
Config -> Devices -> Edit
Port  HC-06 (as BLE Bluetooth Low Energy)
Driver KRT2
K6Bt Off

Status (after connected) shows No Data

The environment is that a HC-06 device is powered and connected via serial interface to a KRT2 radio.  The bluetooth connection and serial interface was independently fully tested on both phones using Android app: "Bluetooth terminal".

MaxKellermann, 2023-05-30:
Post the full verbatim error message extracted via logcat including surrounding log lines and the backtrace. Without this, we cannot analyze your problem.

nrbray, 2023-05-30:
> Post the full verbatim error message extracted via logcat including surrounding log lines and the backtrace. Without this, we cannot analyze your problem.

Dear Max,  

I am happy to follow up with this information but have not done this before so would appreciate some pointers on how to get started.  (I am at the google for what is logcat stage).

A complication may be that the radio is in the glider.  I did not see the error when I tried it out on the bench, but will see if I can repeat the error without the KRT2 connected first.

Best regards Nigel

MaxKellermann, 2023-05-30:
If XCSoar crashes, you can probably extract the full error message hours later when you're back at a computer. On Linux, I use the Android SDK with the command "adb logcat" to extract the Android system log to my computer, but I don't know how to do it with any other operating system. (In any case, I can only help you if I have the full log, no matter how you obtain it.)

MaxKellermann, 2023-07-27:
Still no logcat with backtrace

closed 2023-07-27 by MaxKellermann