XCSoar/XCSoar #1241: Distance calculation

Soaringseven, 2023-06-05:
Dear folks, during my last flights the distance calculator (former OLC classic) suddenly adds 100 km or more which hadn't been flown yet. After some time (30 min), the systems recalculates the distance back to the real level.  I use a Xiomi 9c Redmi. In the past, the same smartphone never showed problems with XCsoar. Is it a problem of my smartphone or of the new XCsoar-version? Best regards Holger

MaxKellermann, 2023-06-05:
Nobody can know because you didn't post enough information. We need at least a NMEA/IGC file which can be used to reproduce the problem in replay. Upload one, but first verify that it can indeed be reproduced.

Soaringseven, 2023-06-08:
Hi Max,

this is the igc-file produced by XC soar. The distance-skip of about 100 km
happens after km 170.

Mit freundlichem Gruß

Holger Linnartz
Bahnstr. 24a
D- 47929 Grefrath
0174-1730570
02158-4059069

MaxKellermann, 2023-07-18:
There's no IGC file.

MaxKellermann, 2023-07-27:
No reply

closed 2023-07-27 by MaxKellermann