XCSoar/XCSoar #1242: PEV not send to LXNAV nano

kobedegeest, 2023-06-06:
XCSoar versions having and not having the problem
-------------------------------------------------
Only tested current version 7.31

Expected behavior
-----------------
Pressing Pilot Event on XCSoar sends PEV to al connected loggers. 

Actual behavior
---------------
Pressing PEV is not send to (or not received by) LXNAV nano which is connected via bluetooth and working fine. 

Steps to reproduce the behavior
-------------------------------
Connect an LXNAV nano to xcsoar with bluetooth fly and press pev during flight check nano igc log file. 
Device is configured in xcsoar as device A with driver LXNAV, passthrough device off, sync from device on sync to device on K6Bt off.
Details phone:
Model: Xiaomi Redmi Note 7
Bluetooth version: 5.0 
Android version: 10

Attached log file xcsoar, nmea file xcsoar, log file nano.
[logfilesnopev.zip](https://github.com/XCSoar/XCSoar/files/11663932/logfilesnopev.zip)
I have more files of following days where i pressed the pev button on nano first and then on xcsoar every day only one pev (from nano itself) is registered in the log. 

If you need more info to be able to debug this just let me know.

Do you have any idea what may have caused this?
-----------------------------------------------
No.
I do use a custom xci file which has pev button easier accesable but since it does everything correct on xcsoar i do not think that is the issue but in annycase here is what i have in the xci file for pev button.

mode=CMup1
event=Mode default
event=StatusMessage Pilot event announced
event=Logger note PEV
event=PilotEvent
label=PEV
location=3

Do you have an idea how to solve the issue?
-------------------------------------------
no.

Scumi, 2023-06-06:
Last I checked with LXNAV, the Nano does not support reception of external commands to trigger PEV.

Please contact LXNAV to check. I can quickly create a PR to add this once it is available.

MaxKellermann, 2023-06-06:
Since this is currently impossible to implement due to external constraints (not supported by the logger you have), there's nothing we can do, so I'll close this feature request. We can reopen it once LXNAV implements this in a firmware update and publishes the protocol documentation.

closed 2023-06-29 by MaxKellermann