XCSoar/XCSoar #1243: Feature request: QuickMenu and Map button to toggle Airspace Labels

DanD222, 2023-06-09:
What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
At present, airspace labels on the map can be toggled on or off only from the Settings (Map Display / Airspace / Label visibility), which I believe triggers `AirspaceLabelSelection`. 

Since the airspace labels take up quite a lot of space on the map, it would be useful if their display could be toggled quickly, rather than by having to navigate into the settings.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

For this purpose, a QuickMenu button to toggle airspace labels would be useful, as well as button under the Display button for easier access on mobile devices.

lordfolken, 2024-06-02:
@DanD222 this should be easy to implement. see Data/Input/default.xci, want to have go?

lordfolken, 2025-12-28:
Added to both quick menu and button menu.

closed 2025-12-28 by lordfolken