XCSoar/XCSoar #1244: NMEA line endings

pjalocha, 2023-06-09:
XCSoar version
--------------

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

lordfolken, 2023-06-09:
So how did you resolve this problem with this barometer?

pjalocha, 2023-06-09:
I fixed the ESP32 code which due to some line-ending setting in sdkconfg (which somehow went in there when playing with different ESP-DF versons) was producing LF+LF and then it worked but it took me some time to find out the reason, so if the line ending where shown in the monitor mode it would attract my attention immediately.

pjalocha, 2023-06-09:
Monitor functon is to debug why things do not work as expected n XCsoar,

so I proposed this small feature.

I do not request it, as I already solved my issue and will well remember it, but there might be a new user trying a new device and he might struglle with it like I did, so this small thing would help it.

MaxKellermann, 2023-07-01:
I don't quite get what you want XCSoar to do. Be "more tolerant", but what does that mean, technically?

pjalocha, 2023-07-01:
For example do not require strict NMEA termination by CR-LF but any of these characters or any combination.
Any type of line termination is clear enough to signal the end-of-line.

or, if not, then signal that the NMEA's are incorrect, so that the user knows right away that something is wrong,
because otherwise the user sees correct NMEA in the monitor but they whole thing does not work,
which is pretty confusing, right ?

kajusK, 2023-07-13:
It would be nice to tolerate any line endings by the NMEA parser, I'm just fighting with XCSoar and Quectel L86G module - can see it has fix in the monitor, but XCSoar keeps telling me there's no GPS connected and the NMEA Output device has no data, this is really CONFUSING. Took me a while to as a google the right question and end up in this thread.

pjalocha, 2023-07-13:
Yes, it looks right but iit is not right... and you don't see why ?

This is because end-of-line logic for the monitor is different from end-of-line logic for NMEA
so the monitor shows the end-of-line but NMEA decoder does not consider it.

Either show the line endings in the monitor or make NMEA end-of-line logic more elastic or both.

MaxKellermann, 2024-02-28:
XCSoar expects lines to end with LF (but does not require CR, even though the NMEA standards mandates that lines end with CR+LF).
I'm not convinced that XCSoar should allow lines to end only with CR. That simply makes no sense, and anybody who emits broken lines like that should fix their code.
Nobody here has explained why supporting CR-only lines is necessary or useful.

closed 2024-02-28 by MaxKellermann