XCSoar/XCSoar #1245: XCsoar on Kobo crash when moving

MmeHulot, 2023-06-09:
XCSoar versions having and not having the problem
-------------------------------------------------
XCSoar v7.34-Kobo

Expected behavior
-----------------

when starting in FLY modus Program start, loads waypoints etc., starts GPS everything ok

Actual behavior
---------------

program starts normal in FLY modus. When not moving  everything is fine. When starting moving after a few seconds XCSoar crashes an starts with home screen 

Steps to reproduce the behavior
-------------------------------

Starting in FLY modus then moving with v > 10 km/h

Do you have any idea what may have caused this?
-----------------------------------------------

seems that the GPS data/interface causes the crash. Trying another Port doesn't help. Monitoring GPS data looks fine. Position is correct

Do you have an idea how to solve the issue?
-------------------------------------------

tried other versions 7.24, 7.10 same behaviour. Checked the SD-Card no errors found. Attached .zip contains two files from XCSoarData/crash

[xcsoar.log](https://github.com/XCSoar/XCSoar/files/11709989/xcsoar.log)
[xcsoar.zip](https://github.com/XCSoar/XCSoar/files/11709994/xcsoar.zip)

ubx, 2023-06-10:
Could not reproduce the problem with the latest from head on my Clara 2E.

- What is your Kobo Model?
- Is it connected with an external power supply?
- What is shown under Devices → Monitor?

lordfolken, 2023-06-11:
Your power must be low and xcsoar seems to be shutting down normally according to your log. Maybe we should add a message to the log when shutting down on certain events.

MmeHulot, 2023-06-12:
no time on weekend....Thanks! 
I use an KOBO Touch, which is connected to a Powerbank trough USB. I use an HC07 (don't remember exactly) GPS Module which is wired direct to the serial Interface of the Kobo. Monitoring this via NMEA (dev/ttymxc0 (9600 Baud) gets normal GPS data. As mentioned the system worked well for years now without changes. My idea was maybe the SD-Card is corrupt an when starting the logger (because of start moving) causes the crash, but setting the logger to manual start made no difference. Also a check of the XCsoarData-Partition brought no errors. 
So I think your are right as you checked the log and it may be a Problem with the (old) Battery now beside being connected to a Powerbank via USB (it seems the connected Powerbank doesn't matter, which is not clear for me).  I will check this by buying a new battery an then give feedback.
Maybe this "problem" could be interesting for others when running a "old" system in this configuration with an "old" battery.  A log for this as you proposed would be helpful

MaxKellermann, 2023-06-12:
Your xcsoar.log shows that you successfully started and quit XCSoar. No crash possible.
Your core dumps are not usable with the official 7.34 builds. Did you do your own build?

MmeHulot, 2023-06-12:
No I am not using my own build. I updated to 7.34 and the core.dumps are from 7.24. (sorry for that)  I try to get some core dumps with 7.34. The log may say that I started XCSoar, correct, but I did not quit XCSoar. It quitted itself after starting to move. That is the problem.  As @lordfolken  said above maybe it is a problem with a "weak" battery.

MaxKellermann, 2023-06-12:
7.24 doesn't match either. Please NEVER report bugs for any version but the most recent one. You're only wasting our time with that.

MaxKellermann, 2023-06-12:
That did work. Though I don't understand how this can crash:
```
#0  0x00018800 in is_simulator () at ./src/Simulator.hpp:25
#1  GetGPSDeviceName () at src/Logger/LoggerImpl.cpp:244
#2  LoggerImpl::StartLogger (this=this@entry=0x945000, gps_info=..., settings=..., asset_number=0x7ebf226c "", asset_number@entry=0x2fd6b0 <boost::json::key_value_pair::empty_> "", decl=...) at src/Logger/LoggerImpl.cpp:278
#3  0x00017cfc in Logger::GUIStartLogger (this=this@entry=0x945000, gps_info=..., settings=..., protected_task_manager=<optimized out>, noAsk=true) at src/Logger/Logger.cpp:90
#4  0x00027aa8 in InputEvents::eventLogger (misc=0x2fe245 "start") at src/Input/InputEventsActions.cpp:449
#5  0x00027194 in InputEvents::processGo (eventid=5) at src/Input/InputEvents.cpp:448
#6  0x000275e4 in InputEvents::processGlideComputer_real (gce_id=<optimized out>) at src/Input/InputEvents.cpp:432
#7  0x0002a14a in InputEvents::DoQueuedEvents () at src/Input/InputQueue.cpp:65
#8  0x0002769e in InputEvents::ProcessTimer () at src/Input/InputEvents.cpp:499
#9  0x00040e10 in CommonProcessTimer () at src/ProcessTimer.cpp:173
#10 ProcessTimer () at src/ProcessTimer.cpp:220
#11 0x00042cd0 in MainWindow::RunTimer (this=0x85f850) at src/MainWindow.cpp:672
```
Disassembly:
```
   0x000187f4 <+112>:	mov	r2, r0
   0x000187f6 <+114>:	cbz	r0, 0x18800 <_ZN10LoggerImpl11StartLoggerERK8NMEAInfoRK14LoggerSettingsPKcRK11Declaration+124>
   0x000187f8 <+116>:	ldrb.w	r2, [r4, #112]	@ 0x70
   0x000187fc <+120>:	eor.w	r2, r2, #1
=> 0x00018800 <+124>:	ldr	r1, [pc, #208]	@ (0x188d4 <_ZN10LoggerImpl11StartLoggerERK8NMEAInfoRK14LoggerSettingsPKcRK11Declaration+336>)
   0x00018802 <+126>:	add.w	r3, r6, #8192	@ 0x2000
   0x00018806 <+130>:	ldr	r0, [r6, #4]
   0x00018808 <+132>:	add.w	r4, r4, #384	@ 0x180
   0x0001880c <+136>:	add.w	r12, r5, #128	@ 0x80
```
`[pc, #208] = 0x0081b5e2`
`[0x0081b5e2] = 0x00000000` (`global_simulator_flag`)

MaxKellermann, 2023-06-12:
Wait...
```
Program terminated with signal SIGILL, Illegal instruction.
```
This means your Kobo has a CPU that doesn't understand certain instructions.
Please upload your `/proc/cpuinfo` here.

MmeHulot, 2023-06-12:
my /proc directory on rootfs is empty.
 I am wondering because the whole system was working for years, without problems

MaxKellermann, 2023-06-12:
`/proc` is only a mount point, it will be mounted at runtime, so `/proc/cpuinfo` can only be extracted from within the Kobo. But you can add `cp /proc/cpuinfo /mnt/onboard/XCSoarData/` to `XCSoarData/kobo/init.sh` - that shell script will be executed at every boot.

The crash could have appeared when we updated the compiler version some time ago. Maybe the new compiler emitted instructions that your CPU doesn't know. When we know your CPU, we can adjust the CPU selection in the compiler to fix it.

MaxKellermann, 2023-06-14:
@Bas1950 please don't spam this bug report with an unrelated problem. If you see a bug in the most recent release, write a new bug report and upload the necessary information. But please do not write a bug report for XCSoar 7.18.

MmeHulot, 2023-06-14:
Is there a way to reset the logbook? 
I deleted all the .igc-Files in the /logs directory (336 Files) to "reset" the Logger Data, but the logbook (in powered off modus) still shows flights (some without duration). The idea was starting the logger with no log  (first one) to test if this changes something.

MaxKellermann, 2023-06-14:
The logbook is `flights.log`, but this for sure has nothing to do with your problem.

MaxKellermann, 2023-06-14:
Side note: I cannot reproduce this problem with a Kobo Mini. No SIGILL crash. `cpuinfo`:
```
Processor	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 159.90
Features	: swp half thumb fastmult vfp edsp neon vfpv3 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x2
CPU part	: 0xc08
CPU revision	: 5

Hardware	: Freescale MX50 Reference Design Platform
Revision	: 50011
Serial		: 0000000000000000
```
This appears to the be same CPU as your Kobo Touch, and this looks like an impossible situation.

MmeHulot, 2023-06-14:
@lordfolken checked another battery: not a power issue. 
Starting logger -> crash

MmeHulot, 2023-06-14:
Solution:
I played with the tasks. Loaded another task. Started the Logger: no crash. Loaded another task: no crash. Loaded the "Crash"-Task: Crash! I can reproduce this behaviour. Did a ride with the bike: no crash if the Non-Crash-Task loaded.
I attach the "Crash"-task. No idea why this happens
[crash.tsk.zip](https://github.com/XCSoar/XCSoar/files/11747357/crash.tsk.zip)

MaxKellermann, 2023-06-14:
Nice, with your task, I can reproduce the crash!

MaxKellermann, 2023-06-15:
This is a stack buffer overflow because the turnpoint name is very long. How embarassing!
https://github.com/XCSoar/XCSoar/blob/a970266b8fb1cd5de57ed748b8840ee20858d2f8/src/IGC/IGCWriter.cpp#L144-L145
This bug affects all targets, but apparently only the Kobo happens to really crash, and confusingly it crashes with SIGILL instead of SIGSEGV. I don't quite understand why.