XCSoar/XCSoar #1247: ACD Protocoll active Freq. change does not send any NMEA commands

XCNav, 2023-06-11:
XCSoar versions having and not having the problem
-------------------------------------------------

7.33

Expected behavior
-----------------

Change active freq. on connected ACD when selected on Airport details page

Actual behavior
---------------

When calling up airport details i can select to change actual and sby freq. Monitoring TX OUT (configured with the ACD  Driver) I can see NMEA Commands for Changing the SBY frequency (e.g. $PAAVC,S,COM,CHN2,121215*25)
When selecting the CHGE ACTIVE FREQUENCY button there is no outgoing NMEA command generated 

Steps to reproduce the behavior
-------------------------------
See answer above

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

Scumi, 2023-06-11:
The Air Control Display does not support that request. You can cross-check that in the installation manual.

XCNav, 2023-06-11:
According to the installation manual Chapter F5.4 I understood that the command line supports the change of active and stby frequency. After rereading the chapter I now understood that they are talking about exchanging/swapping the stby/act freq. when connected to an AirCom Radio. I will ask Air Avionics to chge that. Please keep this issue open.

XCNav, 2023-06-14:
Talked to AirAvionics. Feature is not implemented yet due to initial certification issues, but it will implemented on one of the next major updates. However it is still subject to certification and it may take till end of the year.

Scumi, 2023-06-14:
Yes, I got the same answer earlier this year. I actually have the patch for the driver prepared already. Now we just need to wait for the update.

MaxKellermann, 2023-06-29:
Let's reopen when there's actually something we can implement. Right now, there is nothing.

closed 2023-06-29 by MaxKellermann