XCSoar/XCSoar #1248: LXNAV S80 Vario: Download of Flight List fails

lordfolken, 2023-06-13:
XCSoar versions having and not having the problem
-------------------------------------------------
current head. 

Expected behavior
-----------------
After going to Config->Devices->FlightDownload, be presented with a list of flights to choose from, and to then download. 

Actual behavior
---------------

There is an error: Failed to Download flight list

Setup:
---------

* This is a flarm hooked up to the gps port of an S80 Vario running 9.13 firmware.
* The S80 is connected via the bluetooth stick from LXNAV 
* rfcomm bind to the device to /dev/rfcomm0
* the flarm doesn't have gps lock as i'm indoors
* communication from xcsoar running on a linux computer to the vario works (i get environment, flarm, gps(when outside), baro, airspeed and vario
* Bidirectional communication works: MC/Bal/Bugs go both ways
* Declaration of tasks works, (not always)
* There are 50 Flights in the logbook of the device

The configuration in xcsoar:
[image]

MaxKellermann, 2023-06-13:
Since you did not specify a version that worked, I figure that you never tried this with any other XCSoar version?

lordfolken, 2023-06-13:
I will try some other versions.

lordfolken, 2023-06-13:
There is two things that strike me:
* it seems to stop after the 8th flight
* the starting date seems wrong, always 00.00.1980. (althought this would be a bug in the firmware)

lordfolken, 2023-06-13:
So so far i have been able to trace the problem to ParseLogbookContent in ReadLogbookContent. 

It doesn't return true and there is no further handling of the case.

groundhogxc, 2023-06-21:
Thanks for the PR. By the way: Did you (re)start the LX device very late, like just a minute or so before takeoff? I have seen some funny things happen with "historic" dates in IGC files in this scenario, although did not yet encounter this particular problem.

lordfolken, 2023-06-21:
I think the main thing with my setup, is that the flarm is powered through the s80. So it only starts when the vario starts.

kobedegeest, 2023-06-22:
we had (have) same issue in our club with the s100 of invallid dates the issue (according to LX) is the setting "logger always on" in the LX device if this setting is enabled and it does not have a gps fix at startup the whole log file is wasted because it gets assigned an invalid date.

groundhogxc, 2023-06-22:
Here the FLARM is also powered through the LX. (By the way: Can you control/declare/download the FLARM through the LX via pass-through? That is my dream, but it never worked for me).

I am not sure about the "logger always on" setting; I did have the impression that the problem occurs if the first fix is not recorded a sufficiently long time before takeoff detection or something like that. Because it seemed to be connected with "rebooting" the device very close to takeoff.

lordfolken, 2023-06-23:
I got a beta Firmware from lxnav. Ill try it out.

groundhogxc, 2026-02-15:
> The flight list download has been fixed to handle entries with invalid dates (0000-00-00) that the vario returns for corrupted or empty logbook slots. Previously these caused the download to fail; now they are gracefully skipped.

I remember at least one case where the correct file showed up like this (or maybe in a very similar weird way, it had a "corrupted" date). So I did have to get that file out. It was one of these cases where the GPS had acquired the date too close to the takeoff or something like that.

Of course in those cases there is still the SD card route, I guess...

> @lordfolken — this was your own issue, already verified during development.

Hmm, talking to yourself? :)