XCSoar/XCSoar #1252: LXNav devices  - Bluetooth flight download / declaration intermittent

MikeBravo15, 2023-06-19:
XCSoar versions having and not having the problem
-------------------------------------------------

Current version 7.35, run on Pixel 4a. Tested on 2 pcs Nano3 and one Nano4 loggers

Expected behavior
-----------------

XCSoar should be able to download series of flight/declare without issues

Actual behavior
---------------

Currently I can sually download first flight without troubles, but if I select to download next flight it usualy ends with failed download. After several failed atttempts it can start to work again. Sometimes download fails during the data transmit/task declaration. Sometimes it fails even with first flight. LXNav NanoConfig sw, running on the same phone can download series of flights rock solid

Steps to reproduce the behavior
-------------------------------

Connect LX Nano3/4 via BT and try to download series of flight records from the logger

Do you have any idea what may have caused this?
-----------------------------------------------
No

Do you have an idea how to solve the issue?
-------------------------------------------

No

lordfolken, 2023-06-20:
Can you paste they device configuration page, as you are currently using it? also activate debug in the devices and send the log.

groundhogxc, 2023-06-21:
I have seen a problem that might be the same one, with the S80/S100 series. But it might also be different, because:
- Declaration (so far) has always worked, 
- downloading the flight list also always works (so far),
- but: downloading flights fails sometimes. For very long flights (say, 9 hours), it fails almost certainly, for shorter flights it often works. If it fails, sometimes it works on second or third try; sometimes it seems to fail reliably, so I give up and go searching for the sd card.

After a fail, the data connection to the LX device no longer works. Simple "reconnect" in the device menu does not help. One way to get it back is to disable bluetooth in Android (in XCS the device status then says "bluetooth disbaled"), then re-enable bluetooth in Android. The connection then does not come back automatically, but clicking the "reconnect" button brings it back, and you can try again.

I think restoring the connection also worked by restarting XCSoar without disabling/enabling bluetooth, but here I am not 100% sure if I remember that correctly (no access to LX device currently).

Device: Nexus 7, running Android 11.

Sundown3867, 2023-06-21:
Hallo all,
enclosed please find the screenshot of the Device setting and XCSoar log. 
[image]
[xcsoar.log](https://github.com/XCSoar/XCSoar/files/11821488/xcsoar.log)

When I read the log I noticed a lot of Skylines errors - I have SkyLines traffic mode selected to Symbol and Name in the Map Display -> Elements. I turned it off and I could download 6 flight in row without sigle hicckup

bbaer1955, 2024-01-08:
@lordfalken See latest logcat supplied for #1322 which is very similar if not the same issue.

MikeBravo15, 2026-03-09:
I tested the Android version on my Pixel6a from the link above and download is still failing on my Nano. List of flights is displayed OK, but download is intermittent. I found problems especially on larger files (longer flights) - download do not starts at all or fail during the transfer. Recently I also tested the kobedegeest's version #1813 and there download was rock solid