XCSoar/XCSoar #1253: symbols of landable waypoints don't change color to reflect reachability

bomilkar, 2023-06-19:
XCSoar versions having and not having the problem
-------------------------------------------------
The reachability of a landable waypoint on the map is no longer indicated by a green (or orange) ring.
The manual describes the color coding in chapter 4.5 Waypoints.

The issue was introduced between 7.33 and 7.34
It appears on several (all?) platform.

Expected behavior
-----------------

The color coding of landable waypoints should follow the documentation. See chapter 4.5 "Waypoints".
The reachability has 3 levels: unreachable, marginal, reachable. The waypoint symbols should change according to the table.

Actual behavior
---------------
The symbols of landable waypoints don't change to reflect rechability.

Steps to reproduce the behavior
-------------------------------
Run XCSoar in simulation mode and increase the altitude until certain landable waypoints become reachable. Observe the map symbols of these landable waypoints to see if the symbols change color. In 7.34 they don't!

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

MaxKellermann, 2023-06-19:
Can't reproduce.

bomilkar, 2023-06-19:
Please set:
Config -> "Map Display" -> Waypoints -> "Landable symbols" = "Purple circle"
Please read the text to the option and see if increasing the altitude effects the style of the nearby landable waypoints:
"If the waypoint is reachable a bigger green circle is added behind the purple one." 

I compared this:
https://download.xcsoar.org/releases/7.33/LINUX/xcsoar_7.33_amd64.deb
vs.
https://download.xcsoar.org/releases/7.34/LINUX/xcsoar_7.34_amd64.deb

7.33 is OK, 7.34 shows the issue. Comparing the PC ports exhibit the issue, too.

bomilkar, 2023-06-19:
Here are example screen shoots:
From 7.34
[image]
From 7.33
[image]

bomilkar, 2023-06-26:
After looking into this issue a bit more I found:
`Config -> "Glide Computer" -> Route -> "Reach mode" = Off`
**works as expected.**

But
`Config -> "Glide Computer" -> Route -> "Reach mode" = Straight`
or
`Config -> "Glide Computer" -> Route -> "Reach mode" = Turning`
**exhibits the issue.**

kobedegeest, 2023-06-26:
version 7.35 works fine for me on android. 
I see they are colord red/orange in your picture i can only get that if i put some value for the terain height in the settings 
config-system-glide computer-safety factors 
So the waypoint is reacable but not with the desired ground clearens so maybe check that setting.

kobedegeest, 2023-06-27:
I ussaly have it at straight but do not have mountains around so it plays no big role. 
But i tested all three options all behave the same. I did notice that when changing from "off" to one of the other options it needs a second or two to recalculate at that time it also shows them red/orange. 

Only way i can recreate your screen shot is to set the terain heigth to more than the calculated arrival heigth. That terain heigth is the minimum distance you want between you and the ground, it is to safely pass over mountains etc. Manual also states that the marginal case (red/orange color) is when you in theorie can get there but a mountain or something is obstructing you path.

Terain heigth setting plays also no role if you turn the reach mode to off as it does not take the terain into account then. So it would explain what you see perfectly if you have that set to 500m or so.

bomilkar, 2023-06-27:
Really strange. Thanks anyway for your efforts!

One more attempt: start with a "NEW" profile. Set map, waypoints, airspace files. Then set a "New Home".
It positions the glider symbol at Home. I can increase the GPS Altitude to any value I want but Home (right under the glider symbol) never becomes reachable. Although the map label shows the plausible arrival height.

BTW: I noticed that "Reach display" when set to "Terrain shade" also doesn't work as expected for versions > 7.33

MaxKellermann, 2023-07-01:
Still can't reproduce, not even with a new profile.

bomilkar, 2023-07-02:
Here is another attempt to simplify the testcase.
Create an empty ~/.xcsoar
insert the attached PRFs into this directory

[NOT_OK.prf.TXT](https://github.com/XCSoar/XCSoar/files/11929157/NOT_OK.prf.TXT)
[OK.prf.TXT](https://github.com/XCSoar/XCSoar/files/11929159/OK.prf.TXT)
Rename the *.prf.TXT to *.prf

insert a map file: "wget https://download.xcsoar.org/source/map/region/GER_HighRes.xcm"

Run xcsoar 7.35 (from the download area) in Sim mode.
Increase "Alt GPS" until you are 1000 above ground: "H AGL" >= 1000

You should see "Eisenach Kindel" in 5.8 km 330°.

As you increase "Alt GPS" you should see the waypoint symbol changing color when it becomes reachable.
Version 7.35 of xcsoar behaves different depending on which of the 2 PRFs are selected.
But in this testcase it should behave the same, which it does for 7.33 and older.

The difference between the 2 PRFs:

NOT_OK.prf
Config -> "Glide Computer" -> Route -> "Reach mode" = Straight
(Straight is the default setting)

OK.prf
Config -> "Glide Computer" -> Route -> "Reach mode" = Off

I can't see any reason, why in this particular case "Reach mode" changes the behavior.

Scumi, 2023-07-09:
I just flew two days with 7.36 and can confirm this bug. Not happening with the symbols in the alternates dialog though.

hjr, 2023-07-15:
Tried to reproduce with your profiles on a fresh build (Linux), but cannot. All working fine.

lordfolken, 2023-07-24:
So i can reproduce this on win64:

Blank xcsoar profile 
start in sim:
-> get the following site files:
ALPS_HighRes.xcm
CH-ASP-National-OpenAIP.txt
CH-WPT-National-OpenAIP.cup

* Load them in site files
* set the a home airport
* takeoff in the simulator
* set height to higher than final glide

Its more evident when you set:
config-system-waypoints-landable-symbols: traffic lights

There should be: green if reachable, orange if blocked and red if not reachable at all.

Main window:
[image]

Alternates window:
[image]

hag3fe, 2023-10-25:
Just a hint: in version 7.38 the range of operation isn't shown anymore. I tried all parameters.

lordfolken, 2023-10-26:
What do you mean?

bomilkar, 2023-10-26:
I guess you can see it on the 2 screenshots I posted on June 19. The one labeled "From 7.33" has a polygon (near Gammelsdorf) which separates the "range of operation" form the unreachable which has a dotted fill over the terrain.
The relevant parameter to switch this feature on and off is
Config->System->"Glide Computer"->Route->"Reach mode".

hag3fe, 2023-10-26:
In version 7.38 of win_64 the range of operation isn't shown anymore. In 7.31 it is available. I think that there is a connection to the wrong visualisation of airfields in this issue. My suspicion is: something is wrong wiht the calculation of terrain altitude.
As written: just a hint.

bomilkar, 2023-12-27:
It has been **fixed in the 7.40** production release and all later builds up to now.
Thanks!

bomilkar, 2024-05-21:
Now this issue popped up again. Maybe related to the fix to https://github.com/XCSoar/XCSoar/issues/1424 ?

hag3fe, 2024-05-21:
Hello,

as fas as I see, XCSoar changes the color of reachable airfield only to
yellow, if the airfield is hidden by terrain. If the Mc-setting is to
high the color alternates between red and green. My solution is to watch
the bar of goal hight.

Kind regards

Guido Hasel

bomilkar, 2024-05-21:
> Hello, as fas as I see, XCSoar changes the color of reachable airfield only to yellow, if the airfield is hidden by terrain. If the Mc-setting is to high the color alternates between red and green. My solution is to watch the bar of goal hight. Kind regards Guido Hasel

Are you testing a build generated less than 2 days ago? After this https://github.com/XCSoar/XCSoar/pull/1432

lordfolken, 2024-05-21:
I cannot reproduce this anymore with the current head.
The only thing that bothers me here, is that orange marked airfields show as green in the alternates dialog.
[image]

Scumi, 2024-05-22:
Issue is still apparent on the UNIX target (tested on my laptop and on the OpenVario) with the current head f10ff9c6c4.

https://github.com/Openvario/meta-openvario/issues/375

lordfolken, 2024-05-22:
Please update the ticket header with steps to reproduce, starting with a clean XCSoarData. I cannot reproduce this bug.

Scumi, 2024-05-22:
I'm not the owner of the ticket, so I cannot do an update. 

Anyway, I found the first bad commit with the help of `git bisect`: 0ec0b27d38d527ef9e2fd7b01fad8746a34e11de. Before this commit it works as does the reachable line / terrain shade (see #1452)

@bomilkar Can you check if it works for you when compiling 424829ef592cef83890ce6ceff57e10fecef79e4, which is one commit before the bad one?

bomilkar, 2024-05-22:
I noticed: the behavior is platform dependent: the issue appears on a Pi4 but not on a "normal" debian AMD build.
@Scumi The reachable line does NOT show on the Pi4 build.

bomilkar, 2024-05-22:
> Please update the ticket header with steps to reproduce, starting with a clean XCSoarData. I cannot reproduce this bug.

Done.

Scumi, 2024-05-22:
> I noticed: the behavior is platform dependent: the issue appears on a Pi4 but not on a "normal" debian AMD build. @Scumi The reachable line does NOT show on the Pi4 build.

My laptop also runs on linux amd64, so that's not the common denominator. Have you tried compiling the commit I mentioned for the Pi4?

bomilkar, 2024-05-22:
`git checkout 424829ef592cef83890ce6ceff57e10fecef79e4` fixes the issue on Pi4.

Scumi, 2024-05-22:
Well, that means we at least found the commit that causes the issue for some of us. 

Specifically reverting 
```
  constexpr bool operator==(const FlatGeoPoint &) const noexcept = default;
```
to
```
  constexpr
  bool operator==(const FlatGeoPoint &other) const noexcept {
    return IntPoint2D::operator==(other);
  };
```
from the commit mentioned fixes the issue, but I have no idea yet why. Looked through the code for a while now, but my understanding of that part of the code base (and C++ in general) is pretty bad.

Scumi, 2024-05-23:
@bomilkar and I troubleshooted a bit and it seems the issue is caused by using an old compiler. I've updated to gcc-12 and g++-12 and the problem is gone on my linux laptop. The Openvario build probably also runs on an old compiler, so it makes sense that it has the same issues.

lordfolken, 2024-08-29:
This was fixed in v7.43

closed 2024-08-29 by lordfolken