XCSoar/XCSoar #1258: Flight duration is computed as 00:00 on Kobo Mini

dingetje, 2023-06-29:
XCSoar versions having and not having the problem
-------------------------------------------------

V7.27-Kobo
Unknown

Expected behavior
-----------------

Correctly computed flight duration in logbook when Kobo is powered off

Actual behavior
---------------

Duration is computed as 00:00 despite having correct start and landing times in the flights.log file

Steps to reproduce the behavior
-------------------------------

1. make sure in XCSoar Config > Logger > Log book is On
2. Go Fly
3. After landing, shutdown XCSoar and power down Kobo
4. Start and landing time are correctly shown, yet Duration is computed as 00:00

Do you have any idea what may have caused this?
-----------------------------------------------

No idea, this feature has worked in earlier versions.

Do you have an idea how to solve the issue?
-------------------------------------------

Probably a bug was introduced and no unit tests is implemented for this feature

MaxKellermann, 2023-06-29:
Upload a screenshot and your flights.log and an IGC file recorded by XCSoar of a flight which shows 00:00

dingetje, 2023-06-29:
[image]

[flights.log](https://github.com/XCSoar/XCSoar/files/11906210/flights.log)

[2023-06-25-XCS-AAA-02.zip](https://github.com/XCSoar/XCSoar/files/11906225/2023-06-25-XCS-AAA-02.zip)

MaxKellermann, 2023-06-29:
I just only noticed you're 8 releases behind. I couldn't imagine that anybody would ever report a bug for an outdated version. I think you should first update to the latest version, and then confirm that the same problem exists there.

dingetje, 2023-06-29:
I've just updated to the latest version and the problem persists, in fact it's even worse with an empty line showing at the end.

dingetje, 2023-06-29:
Wow, good find!

closed 2023-06-29 by MaxKellermann

MaxKellermann, 2023-06-29:
> it's even worse with an empty line showing at the end

Fixed in 67da82fde5fb243f2a4d0d2e0a53fab9b1d3b145