XCSoar/XCSoar #1260: IOS64 Compilation Failing

HugoKNL, 2023-07-03:
XCSoar versions having and not having the problem
-------------------------------------------------

7.37

Expected behavior
-----------------

XCSoar build for IOS64

Actual behavior
---------------

Crashes on TIFFOpen function not existing. Shows this in terminal:

```
-- Looking for TIFFOpen
-- Looking for TIFFOpen - not found
CMake Error at CMakeLists.txt:175 (MESSAGE):
  Failed to link with libtiff - TIFFOpen function not found
```

Steps to reproduce the behavior
-------------------------------

- Install all dependencies including libtiff 4.5.1 and libgeotiff 1.7.1 (through homebrew) on MacOS 13.3.1 (22E261) on an M1 MacBook Pro.
- Install XCode 14.3.1
- Run ``` make TARGET=IOS64 ipa```

See the compilation fail with error:
```
-- Looking for TIFFOpen
-- Looking for TIFFOpen - not found
CMake Error at CMakeLists.txt:175 (MESSAGE):
  Failed to link with libtiff - TIFFOpen function not found

-- Configuring incomplete, errors occurred!
Traceback (most recent call last):
  File "/Users/hugo/Downloads/XCSoar/./build/thirdparty.py", line 203, in <module>
    x.build(toolchain)
  File "/Users/hugo/Downloads/XCSoar/build/python/build/project.py", line 80, in build
    self._build(toolchain)
  File "/Users/hugo/Downloads/XCSoar/build/python/build/cmake.py", line 122, in _build
    build = self.configure(toolchain)
            ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/hugo/Downloads/XCSoar/build/python/build/cmake.py", line 118, in configure
    configure(toolchain, src, build, configure_args, self.env)
  File "/Users/hugo/Downloads/XCSoar/build/python/build/cmake.py", line 100, in configure
    subprocess.check_call(configure, env=env, cwd=build)
  File "/opt/homebrew/Cellar/python@3.11/3.11.4_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '/Users/hugo/Downloads/XCSoar/output/src/libgeotiff-1.7.1', '-DCMAKE_TOOLCHAIN_FILE=/Users/hugo/Downloads/XCSoar/output/IOS64/lib/build/libgeotiff-1.7.1/cmake_toolchain_file', '-DCMAKE_INSTALL_PREFIX=/Users/hugo/Downloads/XCSoar/output/IOS64/lib/aarch64-apple-darwin', '-DCMAKE_BUILD_TYPE=release', '-GNinja', '-DWITH_UTILITIES=OFF', '-DBUILD_SHARED_LIBS=OFF']' returned non-zero exit status 1.
make: *** [output/IOS64/lib/stamp] Error 1
```

And CMakeError.log says:
```
Determining if the function TIFFOpen exists failed with the following output:
Change Dir: /Users/hugo/Downloads/XCSoar/CMakeFiles/CMakeTmp

Run Build Command(s):/opt/homebrew/bin/ninja cmTC_70c84 && [1/2] Building C object CMakeFiles/cmTC_70c84.dir/CheckFunctionExists.c.o
[2/2] Linking C executable cmTC_70c84
FAILED: cmTC_70c84 
: && /usr/bin/clang --target=aarch64-apple-darwin -Os -g -ffunction-sections -fdata-sections -fvisibility=hidden  -miphoneos-version-min=10.0 -arch arm64 -target aarch64-apple-darwin -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS.sdk -isystem /Users/hugo/Downloads/XCSoar/output/IOS64/lib/aarch64-apple-darwin/include -DNDEBUG -I./output/IOS64/include -DHAVE_POSIX -DHAVE_VASPRINTF -isystem ./output/IOS64/lib/aarch64-apple-darwin/include -DEYE_CANDY -DXCSOAR_TESTING -isystem /include -DCHECK_FUNCTION_EXISTS=TIFFOpen -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS.sdk -Wl,-headerpad_max_install_names  CMakeFiles/cmTC_70c84.dir/CheckFunctionExists.c.o -o cmTC_70c84  -lz  -lm && :
Undefined symbols for architecture arm64:
  "_TIFFOpen", referenced from:
      _main in CheckFunctionExists.c.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

Do you have any idea what may have caused this?
-----------------------------------------------

Not really.

Do you have an idea how to solve the issue?
-------------------------------------------

No.

lordfolken, 2023-07-04:
Looks like an architecture mismatch.

Can you try compiling with V=2

HugoKNL, 2023-07-04:
> Looks like an architecture mismatch.
> 
> Can you try compiling with V=2

Thanks for the quick reply. You meant using the command ```make TARGET=IOS64 V=2 ipa``` right?

It didn't work. Fails at the same point. Have you maybe got another idea on how to solve this?

lordfolken, 2023-07-05:
Do you have more logs? 
Where is libtiff being searched for?

HugoKNL, 2023-07-05:
**Build run in repo fails**
I now see that the build also fails in the Github Actions Workflow run of this repo. You can see the failing run here: https://github.com/XCSoar/XCSoar/actions/runs/5464676898/jobs/9949871214#step:6:3538

The run  before libtiff was updated to fails with another error at an earlier stage:
https://github.com/XCSoar/XCSoar/actions/runs/5410670123/jobs/9832436492#step:6:2416

**More logs**
Logs are created when running this command: ```cmake /Users/hugo/Downloads/XCSoar/output/src/libgeotiff-1.7.1 -DCMAKE_TOOLCHAIN_FILE=/Users/hugo/Downloads/XCSoar/output/IOS64/lib/build/libgeotiff-1.7.1/cmake_toolchain_file -DCMAKE_INSTALL_PREFIX=/Users/hugo/Downloads/XCSoar/output/IOS64/lib/aarch64-apple-darwin -DCMAKE_BUILD_TYPE=release -GNinja -DWITH_UTILITIES=OFF -DBUILD_SHARED_LIBS=OFF```

This is the actual command that is failing in the build. (so building libgeotiff fails).

Next to the CMakeError.log that I posted earlier, I also have the CMakeOutput.log:
```
The target system is: Darwin -  - aarch64
The host system is: Darwin - 22.4.0 - arm64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/clang 
Build flags: -Os;-g;-ffunction-sections;-fdata-sections;-fvisibility=hidden;;-miphoneos-version-min=10.0;-arch;arm64;-target;aarch64-apple-darwin;-isysroot;/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS.sdk;-isystem;/Users/hugo/Downloads/XCSoar/output/IOS64/lib/aarch64-apple-darwin/include;-DNDEBUG;-I./output/IOS64/include;-DHAVE_POSIX;-DHAVE_VASPRINTF;-isystem;./output/IOS64/lib/aarch64-apple-darwin/include;-DEYE_CANDY;-isystem;/include;
Id flags: -c;--target=aarch64-apple-darwin 

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "CMakeCCompilerId.o"

The C compiler identification is AppleClang, found in "/Users/hugo/Downloads/XCSoar/CMakeFiles/3.23.2/CompilerIdC/CMakeCCompilerId.o"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/clang++ 
Build flags: -Os;-g;-ffunction-sections;-fdata-sections;-fvisibility=hidden;;-miphoneos-version-min=10.0;-arch;arm64;-target;aarch64-apple-darwin;-isysroot;/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS.sdk;-isystem;/Users/hugo/Downloads/XCSoar/output/IOS64/lib/aarch64-apple-darwin/include;-DNDEBUG;-I./output/IOS64/include;-DHAVE_POSIX;-DHAVE_VASPRINTF;-isystem;./output/IOS64/lib/aarch64-apple-darwin/include;-DEYE_CANDY;-isystem;/include;
Id flags: -c;--target=aarch64-apple-darwin 

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "CMakeCXXCompilerId.o"

The CXX compiler identification is AppleClang, found in "/Users/hugo/Downloads/XCSoar/CMakeFiles/3.23.2/CompilerIdCXX/CMakeCXXCompilerId.o"

Detecting C compiler ABI info compiled with the following output:
Change Dir: /Users/hugo/Downloads/XCSoar/CMakeFiles/CMakeTmp

Run Build Command(s):/opt/homebrew/bin/ninja cmTC_ae44a && [1/2] Building C object CMakeFiles/cmTC_ae44a.dir/CMakeCCompilerABI.c.o
Apple clang version 14.0.3 (clang-1403.0.22.14.1)
Target: arm64-apple-darwin
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
clang: warning: -Wl,-v: 'linker' input unused [-Wunused-command-line-argument]
 "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang" -cc1 -triple arm64-apple-ios10.0.0 -Wundef-prefix=TARGET_OS_ -Wdeprecated-objc-isa-usage -Werror=deprecated-objc-isa-usage -Werror=implicit-function-declaration -emit-obj --mrelax-relocations -disable-free -clear-ast-before-backend -disable-llvm-verifier -discard-value-names -main-file-name CMakeCCompilerABI.c -mrelocation-model pic -pic-level 2 -mframe-pointer=non-leaf -fno-strict-return -ffp-contract=on -fno-rounding-math -funwind-tables=1 -faligned-alloc-unavailable -fobjc-msgsend-selector-stubs -target-sdk-version=16.4 -fvisibility-inlines-hidden-static-local-var -target-cpu apple-a7 -target-feature +v8a -target-feature +crypto -target-feature +fp-armv8 -target-feature +neon -target-feature +zcm -target-feature +zcz -target-feature +sha2 -target-feature +aes -target-abi darwinpcs -fallow-half-arguments-and-returns -mllvm -treat-scalable-fixed-error-as-warning -debug-info-kind=standalone -dwarf-version=4 -debugger-tuning=lldb -target-linker-version 857.1 -v -ffunction-sections -fdata-sections -fcoverage-compilation-dir=/Users/hugo/Downloads/XCSoar/CMakeFiles/CMakeTmp -resource-dir /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/14.0.3 -dependency-file CMakeFiles/cmTC_ae44a.dir/CMakeCCompilerABI.c.o.d -skip-unused-modulemap-deps -MT CMakeFiles/cmTC_ae44a.dir/CMakeCCompilerABI.c.o -sys-header-deps -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS.sdk -isystem /Users/hugo/Downloads/XCSoar/output/IOS64/lib/aarch64-apple-darwin/include -isystem ./output/IOS64/lib/aarch64-apple-darwin/include -isystem /include -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS.sdk -D NDEBUG -I ./output/IOS64/include -D HAVE_POSIX -D HAVE_VASPRINTF -D EYE_CANDY -internal-isystem /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS.sdk/usr/local/include -i [... cut ...]
     PUBLIC_HEADER DESTINATION ${GEOTIFF_INCLUDE_DIR}
     ARCHIVE DESTINATION ${GEOTIFF_LIB_DIR} )

# Define grouping of source files in PROJECT file (e.g. Visual Studio)
SOURCE_GROUP("CMake Files" FILES CMakeLists.txt)
SOURCE_GROUP("Header Files" FILES ${GEOTIFF_LIB_HEADERS})
SOURCE_GROUP("Source Files" FILES ${GEOTIFF_LIB_SOURCES})

###############################################################################
# Build GeoTIFF utilities

IF(WITH_UTILITIES)
    ADD_SUBDIRECTORY(bin)
ENDIF()

ADD_SUBDIRECTORY(cmake)
```

lastant, 2023-07-12:
Getting the same on Ubuntu 22.04 (actually WSL) building for Android Arch64

lastant, 2023-07-13:
For some reason libtiff doesn't provide the TIFF_LIBRARIES variable when it's found with FIND_PACKAGE by CMake. What I did to circumvent this, was to edit the actual cmake: added this line to the file `/usr/share/cmake-3.22/Modules/CheckFunctionExists.cmake`:

`set(CMAKE_REQUIRED_LIBRARIES "/home/lastant/XCSoar/output/ANDROID/arm64-v8a/lib/aarch64-linux-android/lib/libtiff.a" -lz -lm)` 

on line 58 (below `macro(CHECK_FUNCTION_EXISTS FUNCTION VARIABLE)`)
The paths need to be adjusted for the specific system of course.

MaxKellermann, 2023-07-27:
12dd3a9ac363c49ac8570c33de6dc4697d800c23

closed 2023-07-27 by MaxKellermann

HugoKNL, 2023-07-27:
Thanks for the update. The build is still failing though. Seen te compile run here: https://github.com/XCSoar/XCSoar/actions/runs/5680052373/job/15393392144#step:7:5314

I'm not experienced with the compiling process, but I'm here to help if you need any info. (sideloading will come to iOS in europe so it'd be nice if it works.)

lastant, 2023-07-27:
Apparently libtiff fixed that issue themselves: https://gitlab.com/libtiff/libtiff/-/issues/589