XCSoar/XCSoar #1264: New FLARM Device Database files not shown in file selection

jirimaier, 2023-07-15:
XCSoar versions having and not having the problem
-------------------------------------------------

Seen in 7.34 on Linux and 7.36 on Android

Expected behavior
-----------------

FLARM Device Databases should be listed in the file selection

Actual behavior
---------------

`GLB-FLARM-DeviceDatabase-FLARMNET`, `GLB-FLARM-DeviceDatabase-OGN`, and `GLB-FLARM-DeviceDatabase-UNITED` don't appear in the list of files

Steps to reproduce the behavior
-------------------------------

Go to Site Files -> FLARM Device Databases -> Download
Download one of the files
Close the Configuration menu
Open the menu again
Go to Site Files -> FLARM Device Databases
The previously downloaded file is not listed

Do you have any idea what may have caused this?
-----------------------------------------------

Unlike the FLARM Device Databases files in older versions, these files have no extension

Do you have an idea how to solve the issue?
-------------------------------------------

Renaming the files by adding `.fln` extension works as a workaround

MaxKellermann, 2023-07-15:
This is a https://github.com/XCSoar/xcsoar-data-content bug, isn't it? @lordfolken?

lordfolken, 2023-07-16:
indeed the extensions are missing. 
```ini
uri=https://download.xcsoar.org/content/waypoint/country/GL-WPT-National-OpenAIP.cup
type=waypoint
description=Greenland aviation data from OpenAIP
update=2023-07-01

name=GLB-FLARM-DeviceDatabase-FLARMNET
uri=https://www.flarmnet.org/static/files/wfn/data.fln
type=flarmnet
description=FlarmNet plane registration data
update=2023-07-01

name=GLB-FLARM-DeviceDatabase-OGN
uri=http://ddb.glidernet.org/download/download-fln.php
type=flarmnet
description=OpenGliderNet plane registration data
update=2023-07-01

name=GLB-FLARM-DeviceDatabase-UNITED
uri=https://turbo87.github.io/united-flarmnet/united.fln
type=flarmnet
description=FlarmNet, OGN and WeGlide live tracking data
update=2023-07-01

name=GLB-WPT-ProvingGrounds-XCSoar.cup
uri=https://download.xcsoar.org/content/waypoint/region/GLB-WPT-ProvingGrounds-XCSoar.cup
type=waypoint
update=2023-07-01

name=GLB-XCI-xcremote-XCNav.xci
uri=https://download.xcsoar.org/content/xci/global/GLB-XCI-xcremote-XCNav.xci
type=xci
update=2023-07-01
```
Fixed in:
https://github.com/XCSoar/xcsoar-data-content/commit/e2f15bd3fcd382ed8a91b607bec26bd68d6c0b60

closed 2023-07-16 by lordfolken