XCSoar/XCSoar #1265: Android app stops when exiting edit task

GuyCorbett170, 2023-07-18:
XCSoar versions having and not having the problem
-------------------------------------------------

Current dev branch 18/07/23 (commit 449499b)
Last time I built was 3 weeks ago without this issue

Expected behavior
-----------------

,Enter task dialog, add or replace a turn point, close dialog. I expect it to save task changes

Actual behavior
---------------

Message box 'XCSoar has stopped X Close App".  On rerruning see task changes not saved

Steps to reproduce the behavior
-------------------------------

build latest dev branch as Android. Run Android App in fly or simulator. Enter task dialog, add or replace a turn point, close dialog.  I tried building as UNIX64 and that did not show this problem.  I have only tried on 1 Android device so far but loading an older version worked properly.  

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

GuyCorbett170, 2023-07-18:
Log file attached but not very helpful

MaxKellermann, 2023-07-18:
Need a logcat to analyze the crash, and bisect the bug.

GuyCorbett170, 2023-07-18:
[xc.log](https://github.com/XCSoar/XCSoar/files/12083642/xc.log)

GuyCorbett170, 2023-07-18:
I'm afraid this is outside my knowledge.  I had not run logcat before and will not to learn about a bisect and how to answer your question.  I will try to learn more over the next few days

GuyCorbett170, 2023-07-19:
You were right. rebuilding the whole output fixed the problem.  Thanks for your help

closed 2023-07-19 by GuyCorbett170