XCSoar/XCSoar #1268: Spontaneous wind display

hjr, 2023-07-22:
XCSoar version 7.36
--------------

XCSoar has no feature roadmap or plan to pick up the opportunities given through recent affordable and precise sensors to generate wind calculation with low latency, 3d models of air flow surrounding the glider, finally making a pneumatic TE compensation needless etc. 
I'd like to see XCSoar supporting this move, but I have to admit the topics is ugly. New devices have non NMEA 0183 protocols and go completely their own way. OpenVario advertises to use  $POV, Wis sentence to transfer instant wind estimates, but I am not sure what a good solution would be.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
Have the instant wind estimate on the XCS display rather than on extra little display toys.

Would like to extend the POV nmea implementation with the POV,Wis ; or also any better suggestion here.

Thx

MaxKellermann, 2023-07-23:
I don't get it. What exactly do you want XCSoar to do? Something with "instant wind" (what is the definition of "instant wind"?), but where does it come from? You want to display "instant wind" on XCSoar, but that's what https://github.com/XCSoar/XCSoar/issues/1269 is, isn't it?
How is https://github.com/XCSoar/XCSoar/issues/1086 related to this feature request?

hjr, 2023-07-27:
compiled it into #1269

closed 2023-07-27 by hjr