XCSoar/XCSoar #1269: Instant wind visualization

hjr, 2023-07-22:
XCSoar version 7.36
--------------

As a sibling to #1268, I'd like to propose an additional second sensor value placeholder for an instant wind estimate, respectively an appropriate visualisation in case it is present.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
Visualize the instant wind estimate on the current XCS display.

Suggestion would be:
1. Same arrow as the current on circing around the glider symbol, but with a different color.
2. Arrow with a different radial position around the glider symbol.
3. Different graphical item/arrow

DanD222, 2023-08-25:
FWIW, here's the LARUS protocol documentation:
https://github.com/larus-breeze/standards-larus-NMEA-protocol

I believe what we are interesting in is parsing the `$PLARW` Wind sentence and displaying the "I" parameter of field `3. (A)verage or (I)nstantaneous wind` as per the visualization suggestions above.

lordfolken, 2024-06-02:
now that we have the larus driver merged. Could someone of you pick this up?

Witloofboer, 2024-06-18:
@August2111 has a fork (https://github.com/OpenSoaring/OpenSoar) that shows instantaneous and average winds. I had added some basic functionality earlier as well. @August2111 do you intend to merge your fork again? Otherwise I'll see if what I have still works with the latest release and maybe cherry pick from the OpenSoar commits.

dspreitz, 2026-11-08:
Current state on master: the Larus driver parses `$PLARW` but only uses the averaged wind. An `I` sentence is dropped (`Larus.cpp:307`), there is no separate field for it in `NMEAInfo`, and the map draws a single wind arrow.

I'd like to pick up step a) from the description: read instantaneous wind from NMEA and show it. Roughly:
- a separate instantaneous wind value in `NMEAInfo`, next to `external_wind`
- the Larus driver provides it from `$PLARW` with `I`, with parser tests
- a second arrow on the map in a different colour, shown only while the value is valid, and probably an InfoBox

Step b), calculating it inside XCSoar from GNSS, airspeed and attitude, would stay out of this. It is a much larger job, and the Larus sources are GPL-3 while XCSoar is GPL-2.0-or-later.

Before I start: @lordfolken, is this the direction you'd want? @Witloofboer @August2111, is anyone working on this, or planning to bring the OpenSoar implementation over? I don't want to duplicate that.

lordfolken, 2026-11-09:
for a:
@dspreitz there is a ready2merge pull request covering a) here:
https://github.com/XCSoar/XCSoar/pull/2954
If you could review it?

August2111, 2026-11-10:
Quick update on a): #2954 has been closed by its author in favour of #3469, which contains the same change split into separate commits (NMEAInfo, Larus driver with test, renderer, look, map). On top it expires the instantaneous wind after 10 s, fixes the TestDriver plan count and draws a measured average wind in its own colour. So #3469 is now the one to review – @dspreitz, if you could take a look there instead, that would be great. Thanks @Lordfolken for pointing to it!

Hi @dspreitz, thanks for asking first – this is already done: #3469 implements exactly step a) as you describe it, based on the OpenSoar implementation that has been in use since 2023 (original Larus part by @Witloofboer, credited as co-author):

NMEAInfo gets external_instantaneous_wind next to external_wind; it expires after 10 s, so the arrow is only shown while the value is valid
the Larus driver passes on $PLARW ... I, with a TestDriver check
the map draws the instantaneous wind as a second, light green arrow; in addition, a measured average wind is drawn light blue instead of the grey of XCSoar’s own estimate

What #3469 does not contain yet is the InfoBox. OpenSoar has two of them (speed and bearing of the instantaneous wind); I’m happy to add them to #3469 or bring them as a follow-up, whichever you prefer.

Agreed on step b): calculating the wind inside XCSoar is a separate, much larger job, and nothing in #3469 comes from the Larus sources.

So a review of #3469 would be the most helpful thing right now.

closed 2026-11-11 by lordfolken