XCSoar/XCSoar #1270: QR code for importing tasks

Persiantraveler, 2023-07-23:
XCSoar version
--------------

 7

What should XCSoar do differently, what functionality should be added?
Adding QR code import feature that allows tasks to be ported using the camera on the device. XCtrack uses this feature and it is a very time saving,error reducing and more accurate way to imprt task into their program. I hope Xcsoar would consider adding this feature too. 
----------------------------------------------------------------------

Described it in above

What are you trying to do, what is the use case for the suggestion? As explained above using QR to import tasks is a very good feature to have an it is very popular feature of XCtrack which could be used in other apps such as xcsoar. 
-------------------------------------------------------------------

MaxKellermann, 2023-07-23:
How does that work, technically? A QR code encodes a string of data. What is inside this string of data? Please post a link to a specification and upload a few samples with data how these samples are supposed to look like in XCSoar.

Scumi, 2023-07-23:
Related: #1080

Persiantravelerr, 2023-07-23:
Basically it works by encoding the task info such as coordinates for waypoints, start and finish and other variables such as SSS and ESS as time stamp into a QR code which then can be shared and imported into relevant apps that pilots use such as XcTrack, Naviter Oudie N, FlySkyHi...etc.

For example this is a task we had recently during the US Open.

As you can see the organizers had to only send that QR code to all participants as a text message and when we scanned that QR through whichever apps each pilot may use, in my case XcTrack, then the app would convert that QR to the Task map and Task way point list including the time start, finish ...etc in a matter of a second. 
Please take a look at these pictures for reference. 

 
[image]

[image]

[image]

Hope the following link and example codes I found online can help to illustrate how XCtrack seem to generate and also use QR code. 

https://pkg.go.dev/github.com/twpayne/go-xctrack

Here is a video in german of how you can create a task on one of the XcPlanner kind of Websites and then generate a QR code and import it into your flight app of choice. 
Please fast forward to minute 8:10 if you prefer to see how the presenter sets his task and you can see by minute 13:10 he has created the QR code. 

So any App that can read QR such as XcTrack will be able to create an exact task like the one in this video by just scanning that QR code. 

https://m.youtube.com/watch?v=n3UJlb-ue_0

Here is another example 
https://m.youtube.com/watch?v=OSSsYmzK1Oo

This is how AlphaPilot app does it:

https://m.youtube.com/watch?v=P0rrO4ZhKi8

Hope I have answered someone your questions.

XCNav, 2023-07-24:
LK8000 has this feature as well implemeso brunottl may be able to help

MaxKellermann, 2023-07-24:
I rather not implement it like LK8000 did.

XCNav, 2023-07-24:
However it is managed later in the code, but this maybe a hint of how the qr coding of gliding related data may work.

Persiantravelerr, 2023-07-24:
So far as end user experience might be if a concern, there is definitely room to improve the way QR is implemented. 

For example the way XcTrack is structured, to get to QR import or share page, you need to go through a few steps. 

A much more intuitive approach would be for XcSoar to make a dedicated tab in task setting page or main menu for the purpose of ease of access and exposure prominence.

elgandoz, 2023-07-25:
There is an official page from XCTrack about the QR code format.
Currently the app uses version 2 in order to reduce characters and therefore generate a simpler QR (for easier readability of the camera):
- https://xctrack.org/Competition_Interfaces.html

Despite this is great and widely used in paragliding comps, I think that format may be limited for "sailplanes" tasks given there are more options in the competition rules and type of waypoints.

SeeYou Navigator (Oudie) and it's planner counterpart SeeYou Cloud also generate such QR, with more waypoint options (like "FAI Sector"), but I reckon their format must be an extension of the XCTrack one given the 2 are somehow compatible.

Unfortunately I could find any tech specs, but it could be possible to find out more by checking the output of their [webapp](https://seeyou.cloud/xcplanner). More info: https://naviter.com/2021/10/scan-qr-code-to-enter-the-task/

MaxKellermann, 2023-07-25:
> Unfortunately I could find any tech specs

Then XCSoar won't support it. I won't waste any of my precious time reverse-engineering this.

Meanwhile, I implemented support for XCTrack tasks, and tomorrow we'll have QR scanner support for this on Android, finishing this feature request. The next release (due this week) will have it.

Persiantravelerr, 2023-07-25:
Thank you all. 
Definitely a great feature addition to XcSoar.

XCNav, 2023-07-25:
Perfect.  Thank you!

closed 2023-07-27 by MaxKellermann

Persiantravelerr, 2023-09-29:
Can someone point me to how access the QR feature and how it can be used. Having hard time to find the feature !

lordfolken, 2023-09-29:
Read the text in the comnit.

elgandoz, 2023-09-29:
> Can someone point me to how access the QR feature and how it can be used. Having hard time to find the feature

Just scan it like any other QR, possibly using  your phone camera, **outside** of XCSoar. Once detected, tap the link and the task should open automatically in XCSoar.
It is actually simpler compared to XCTrack.

mzeddd, 2024-10-01:
Sorry, did not get how to use it.
If I scan it with camera it shows JSON text. What should I do with it?

Other tools like SeeYou, LK, XCTrack have button inside task menu to scan task.

HeroicLemon, 2025-04-19:
> Sorry, did not get how to use it. If I scan it with camera it shows JSON text. What should I do with it?
> 
> Other tools like SeeYou, LK, XCTrack have button inside task menu to scan task.

For @mzeddd and anyone else who had a little trouble with this feature, I had to use the actual Android QR code scanner to get the option to open it with XCSoar. 

On a Pixel 7 with Android 15: 

1. Swipe up from the home screen to bring up the app list
2. Search for "QR" and select `QR code scanner`
3. Scan the QR code (test with the QR code in [this](https://github.com/XCSoar/XCSoar/issues/1270#issuecomment-1646823393) comment first to ensure you are scanning a supported XCTrack QR code)
4. Click the `Open` button on the `Open with XCSoar` element.

My usual method of QR code scanning is to double tap the power button to bring up the camera, and then use the camera's built-in QR code scanner which works fine for links and the like, but does not seem to support sending the data to other apps.