XCSoar/XCSoar #1271: Android: Build-in GPS not accessible

ubx, 2023-07-23:
XCSoar versions having and not having the problem
-------------------------------------------------

**Which XCSoar version are you using when seeing the problem**
Latest from HEAD on Android 8.0.0

**Which XCSoar version is the last one you know that did not show the problem**
Investigation with git bisect:
```
$ git bisect bad
cbc2dc0940aaa984d678dff960e0cee40984f33e is the first bad commit
commit cbc2dc0940aaa984d678dff960e0cee40984f33e
Author: Max Kellermann <max.kellermann@gmail.com>
Date:   Wed Jul 19 13:42:59 2023 +0200

    Android: request ACCESS_FINE_LOCATION only when internal GPS is used
    
    In simulator mode (or when connecting to an external GPS), XCSoar
    doesn't need access to the internal GPS, so don't ask the user to
    allow it, it's pointless.
    
    This commit adds some infrastructure to decouple Activity's permission
    request methods from class InternalGPS.

 NEWS.txt                           |  1 +
 android/src/InternalGPS.java       | 24 ++++++++++++++++--
 android/src/NativeView.java        | 10 ++++++--
 android/src/PermissionManager.java | 20 +++++++++++++++
 android/src/XCSoar.java            | 52 +++++++++++++++++++++++++++++++++++---
 src/Android/InternalSensors.cpp    | 11 +++++---
 src/Android/InternalSensors.hpp    |  3 ++-
 src/Android/Main.cpp               |  4 +++
 src/Android/Main.hpp               |  2 ++
 src/Device/Descriptor.cpp          |  5 ++--
 10 files changed, 119 insertions(+), 13 deletions(-)
 create mode 100644 android/src/PermissionManager.java
```

Expected behavior
-----------------
XCSoar should connect to the Build-in GPS

Actual behavior
---------------
XCSoar  doesn't connect to the Build-in GPS. On the map screen "GPS not connected" and in
```
Config -> Devices 
A: Build-in GSP & sensors 
ERROR
```

Steps to reproduce the behavior
-------------------------------
Fresh installation, start XCSoar in FLY mode.

Do you have any idea what may have caused this?
-----------------------------------------------
I must have something to do with the permission request. In Android, the Location Permission for the App is not set. If I set it manually, the problem is gone.

MaxKellermann, 2023-07-23:
Are you really sure it shows "ERROR"? Exactly written this way? Because XCSoar has no such error message anywhere.
Does the logcat show something related?

ubx, 2023-07-23:
No, but here is a screenshot with the exact message.
[image]

ubx, 2023-07-23:
And the logcat.
[xcsoar-logcat.txt](https://github.com/XCSoar/XCSoar/files/12139010/xcsoar-logcat.txt)

ubx, 2023-07-23:
And also full logcat starting XCSoar.
[full-logcat.txt](https://github.com/XCSoar/XCSoar/files/12139077/full-logcat.txt)

MaxKellermann, 2023-07-23:
Okay, that's "Error" and not "ERROR" .... Being very precise is important!
I'll check the logcat tomorrow.

MaxKellermann, 2023-07-24:
There's no error in the logcat.
When this occurred, what was the location permission setting?

ubx, 2023-07-24:
To be exact, here a screenshots:
[image]

ubx, 2023-07-24:
> Which means you disallowed XCSoar to access the GPS location.

No, I didn't disallow GPS location access. The screenshot was taken after a fresh installation and the first start. 
And XCSoar never asked for GPS access.