XCSoar/XCSoar #1272: F-Droid build failed

linsui, 2023-07-24:
```
make[1]: *** [build/compile.mk:104: output/ANDROID/x86_64/opt/src/net/client/WeGlide/UploadIGCFile.o] Error 1

make[1]: *** Waiting for unfinished jobs....

src/Airspace/NearestAirspace.cpp:41:21: error: no template named 'predicate' in namespace 'std'

std::predicate<const AbstractAirspace &> auto predicate) noexcept

~~~~~^

1 error generated.

make[1]: *** [build/compile.mk:103: output/ANDROID/x86_64/opt/src/Airspace/NearestAirspace.o] Error 1

rm output/ANDROID/x86_64/opt/src/Device/Driver/../dirstamp

make[1]: Leaving directory '/home/vagrant/build/org.xcsoar'

make: *** [build/android.mk:333: output/ANDROID/x86_64/opt/bin/libxcsoar.so] Error 2

rm output/host/tools/GenerateSineTables/../dirstampmake: unlink: output/host/tools/GenerateSineTables/../dirstamp: Not a directory

output/host/tools/GenerateSineTables.o
```

We are using GCC 10.

linsui, 2023-07-24:
Which NDK should I use?

MaxKellermann, 2023-07-24:
r26-beta1 - see https://xcsoar.readthedocs.io/en/latest/build.html#compiling-for-android

linsui, 2023-07-24:
Looks like fdroidserver doesn't support beta versions of NDK yet...

MaxKellermann, 2023-07-25:
Not a XCSoar problem

closed 2023-07-25 by MaxKellermann