XCSoar/XCSoar #1273: Wind on the map.

Persiantravelerr, 2023-07-24:
XCSoar version
--------------

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Adding the wind/Gust reading to the map derived from publicly available weather stations in the ground. 
XcTrack has made this feature possible with  simple work around as described and visually presented here:

https://www.pdcs.ch/aktuell/2020/07/24/xctrack-windstationen-anzeigen/

It would be extremely helpful if XcSoar would provide a similar feature. 
 

To use the already available Weather report feature that is under the Info>Weather menu and allow it to be visible on the map similar to waypoints except with current wind/gust data visible next to them. 

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

To make landing safer and to avoid better prior planning for landing by knowing the wind conditions on the ground 

To include free and near realtime realtime local weather stations basic data such as Wind and gust information we would be able to make better landing approaches or perhaps to seek alternative landing locations than originally planned for safety purposes. 

This feature is already available in XCtrack as the workaround has described by a Swiss pilot in the following link:

https://www.pdcs.ch/aktuell/2020/07/24/xctrack-windstationen-anzeigen/

MaxKellermann, 2023-07-27:
We will certainly not embed an HTML browser into XCSoar which shows some website. You can do that already by splitting the screen on Android and showing both XCSoar and a browser. Duplicating this in XCSoar would be rather pointless.

If you still want this feature (but without embedding a website), tell me where to get raw data from these "publicly available weather stations"?

Persiantravelerr, 2023-07-27:
Totally understand but wonder if 
this API be something that can be of use for this potential application ?

https://weacast.github.io/weacast-docs/

Persiantravelerr, 2023-07-27:
Looks like the ones that are realtime are mostly behind paywall for high volume use. 

https://weatherstack.com

So, the split screen should be good enough. 
Thanks for looking into this.

lordfolken, 2023-07-27:
Aprs would be an interesting source for weather stations.

https://aprs.fi/page/api

It would even allow for location based filtering.

MaxKellermann, 2024-02-28:
> split screen should be good enough.

Closing because no XCSoar feature is necessary after all.

closed 2024-02-28 by MaxKellermann