XCSoar/XCSoar #1274: Notaminfo

GrahamStallard, 2023-07-27:
XCSoar versions having and not having the problem
-------------------------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Steps to reproduce the behavior
-------------------------------

Do you have any idea what may have caused this? It seems to be a long line. Deleteing the entery from the file allows the file to load.
Error message attached.
[image]
 With these lines removed
AC NOTAM
AN NOTAM Overflying restrictn misc 26Jun 08:30-25Aug 16:30 Sched:JUN 26-30 0730-1530, JUL 03-07 0730-1530, 10-14 0730-1530, 17-21 <br>0730-1530, 24-28 0730-1530, JUL 31-AUG 04 0730-1530, 07-11 <br>0730-1530, 14-18 0730-1530, 21-25 0730-1530 26Jun 08:30-25Aug 16:30 HIGH INTENSITY RADIO TRANSMISSION AREA (HIRTA) B2011 ACTIVE AT PSN 510939N 0014522W BOSCOMBE DOWN WI 5NM RADIUS MIL ACFT SUSCEPTIBILITY VERY LOW  0.040NM        201FT AGL/576FT AMSL LOW       0.070NM        404FT AGL/779FT AMSL MEDIUM    0.120NM       691FT AGL/1066FT AMSL HIGH      0.320NM      1929FT AGL/2304FT AMSL PULSE LOW/MEDIUM/HIGH   0.00NM          0FT AGL/375FT AMSL THIS ACTIVITY IS WHOLLY CONTAINED WITHIN THE ATZ/MATZ
AL SFC
AH 2407ALT
DP 51:09:39 N 001:45:22 W

the file loads.
-----------------------------------------------

Do you have an idea how to solve the issue? I use 7.24
-------------------------------------------

[notams.20230725.txt](https://github.com/XCSoar/XCSoar/files/12184314/notams.20230725.txt)

MaxKellermann, 2023-07-27:
Your airspace file is obviously broken.

closed 2023-07-27 by MaxKellermann

GrahamStallard, 2023-07-27:
Thank you for the prompt response.

Unlikely as it works with 7.24 then fails using 7.25. Something changed at 7.25. Same file!

 

 

Best Regards,

 

Graham Stallard.

 

Mobile 07801 387154

MaxKellermann, 2023-07-27:
That's a fallacy.

GrahamStallard, 2023-07-27:
And yet it works as downloaded  with 7.24. I am puzzled how it can be a fallacy.

 

It seems I may have caught on a bad day. I thought I was providing feedback to support XCSoar. 

 

 

 

Best Regards,

 

Graham Stallard.

 

Mobile 07801 387154

MaxKellermann, 2023-07-27:
No, it doesn't work with 7.24. That's part of the fallacy you got yourself into.

GrahamStallard, 2023-07-27:
Hello Max,

 

            OK. As you are the main man you would know. As a user when I have 7.24 loaded the files load into XCSoar and XCSoar displays the Notams for the day.

 

            Apologise for being a nuisance.

 

Best Regards,

 

Graham Stallard.

 

Mobile 07801 387154

MaxKellermann, 2023-07-27:
If you insist that you are right and I am wrong, show me a screenshot of XCSoar 7.24 displaying the NOTAM "Overflying restrictn misc 26Jun 08:30-25Aug 16:30 Sched:JUN 26-30 0730-1530, JUL 03-07 0730-1530, 10-14 0730-1530, 17-21".

GrahamStallard, 2023-07-27:
Hello Max,

 

            I attached a screen shot in my case and a copy of the text I cut out after which the file loaded in 7.37. text below.

 

This is the section

AC NOTAM
AN NOTAM Overflying restrictn misc 26Jun 08:30-25Aug 16:30 Sched:JUN 26-30 0730-1530, JUL 03-07 0730-1530, 10-14 0730-1530, 17-21 
0730-1530, 24-28 0730-1530, JUL 31-AUG 04 0730-1530, 07-11 
0730-1530, 14-18 0730-1530, 21-25 0730-1530 26Jun 08:30-25Aug 16:30 HIGH INTENSITY RADIO TRANSMISSION AREA (HIRTA) B2011 ACTIVE AT PSN 510939N 0014522W BOSCOMBE DOWN WI 5NM RADIUS MIL ACFT SUSCEPTIBILITY VERY LOW 0.040NM 201FT AGL/576FT AMSL LOW 0.070NM 404FT AGL/779FT AMSL MEDIUM 0.120NM 691FT AGL/1066FT AMSL HIGH 0.320NM 1929FT AGL/2304FT AMSL PULSE LOW/MEDIUM/HIGH 0.00NM 0FT AGL/375FT AMSL THIS ACTIVITY IS WHOLLY CONTAINED WITHIN THE ATZ/MATZ
AL SFC
AH 2407ALT
DP 51:09:39 N 001:45:22 W

 

The unedited file is in the submitted case.

 

 

Best Regards,

 

Graham Stallard.

 

Mobile 07801 387154

MaxKellermann, 2023-07-27:
There's no screenshot.

GrahamStallard, 2023-07-27:
Hello Max,

 

            I have now actually read your email. I have created a number of Notam files. One with this Notam in question alone. Another with a few Notams that display and a large area Notam that requires a ack..

 

            I take my hat off to you. The Notam alone does not display!. To check I did not have an error in the syntax I added it to a cut down file mid file. Still no display.

 

            I like a puzzle so will try and workout the error in the Notam file for my personal satisfaction.

 

            Apologise for taking up your time. I know you just know how I have been caught out. Have a good evening.

 

 

Best Regards,

 

Graham Stallard.

 

Mobile 07801 387154

GrahamStallard, 2023-07-27:
:) you are correct. The email should have just received puts it in context. 

 

 

Best Regards,

 

Graham Stallard.

 

Mobile 07801 387154