XCSoar/XCSoar #1275: Bluetooth devices not visible in device Port list

DanD222, 2023-07-27:
XCSoar versions having and not having the problem
-------------------------------------------------

Which XCSoar version are you using when seeing the problem?
7.37, 7.38

Which XCSoar version is the last one you know that did not show the problem?
7.36

Steps to reproduce the behavior
-------------------------------
Pair a Bluetooth device (e.g. XCVario) to an Android phone
Start up XCSoar, navigate to Config / Devices
Select an unused device position (A-F)
Select 'Edit'
Select 'Port'

Expected behavior
-----------------
Bluetooth devices should displayed among the listed devices (e.g. Built-in GPS & sensors, TCP client, TCP port, etc).

Actual behavior
---------------
Bluetooth devices are not displayed in the port list.

Devices previously paired and connected to XCSoar function normally, i.e. are displayed in the Devices list (A-F) and XCSoar receives their data. They are however not displayed in the Ports list.

Do you have any idea what may have caused this?
-----------------------------------------------
Recent commits related to 'request BLUETOOTH_CONNECT and BLUETOOTH_SCAN only when Bluetooth is used'?

MaxKellermann, 2023-07-27:
What permissions does XCSoar have on your phone?

DanD222, 2023-07-27:
Location, Storage

MaxKellermann, 2023-07-27:
What setting do you have for Bluetooth? (nearby devices)

MaxKellermann, 2023-07-27:
No, that's neither specific to XCSoar nor is it a permission. It's a global setting.

MaxKellermann, 2023-07-27:
I found it. This old Android version doesn't know permissions BLUETOOTH_CONNECT and BLUETOOTH_SCAN and thus denies them; that makes XCSoar believe the user has explicitly denied Bluetooth permission and doesn't even try to scan. I need to skip those tests on older Androids.

DanD222, 2023-07-27:
Not sure what to check then - I basically try to enable everything for XCsoar. 

Just tried a fresh install, the only permissions I was asked for were:
- Photos, media, files
- Location
 
 I wasn’t asked for any Bluetooth-specific settings for XCSoar, and I have XCVario paired with my phone

closed 2023-07-27 by MaxKellermann

DanD222, 2023-07-27:
Thanks, that was super-quick!

DanD222, 2023-07-28:
Confirming retest OK on 7.39 - thanks again.