XCSoar/XCSoar #1278: XCSoar crashes when I am online during the flight

Blaubart, 2023-07-31:
I have the problem from time to time that my XCSoar (including all commits up to May 26, 2023) crashes on my OpenVario. I then get back to the main menu and can restart XCSoar. 
This always happens when I go online with a WiFi or UMTS USB stick. After a while, XCSoar then crashes in flight and has to be restarted

MaxKellermann, 2023-07-31:
Without a backtrace, this bug report is completely useless.

Blaubart, 2023-07-31:
How can I get the backtrace?

MaxKellermann, 2023-08-01:
Obtain a coredump (e.g. via systemd-coredump) and then use gdb to extract the backtrace. Or let the process crash while gdb is attached.
A bisect would also be very very helpful.

lordfolken, 2023-09-22:
No activity. Closing.

closed 2023-09-22 by lordfolken