XCSoar/XCSoar #1279: Problems with Bluetooth connection on Xiaomi Redmi Note 10

erikkurtz, 2023-08-03:
XCSoar versions having and not having the problem
-------------------------------------------------

7.39
Not sure; might be an issue of or change made by Xiaomi

System information
------------------

Xiaomi Redmi Note 10; MIUI Global 14.0.5; Android 12 SKQ1.210908.001

Steps to reproduce the behavior
-------------------------------

The following error occurs, when trying to connect to any Bluetooth-device (tried with soartronic and AeroFlash):
“UID 10268 / PID 5440 lacks permission android.permission.BLUETOOTH”
(The number behänd “PID” is changing; now it is “14274”)

Expected behavior
-----------------

The connection should work again ;-)

Actual behavior
---------------

No connection possible

Do you have any idea what may have caused this?
-----------------------------------------------

Bruno Tagliapietra mentioned it is caused by a bug in XIAOMI software (I’m not sure if bug or just change).
Here a possible solution is described: https://stackoverflow.com/questions/71552331/got-lacks-permission-android-permission-bluetooth-on-specific-device-after-and

Do you have an idea how to solve the issue?
-------------------------------------------

https://stackoverflow.com/questions/71552331/got-lacks-permission-android-permission-bluetooth-on-specific-device-after-and

MaxKellermann, 2023-08-03:
So it's a Xiaomi firmware bug. Complain to Xiaomi and, most importantly: remember this bug (and their support) the next time you choose a mobile phone.

mlep, 2023-08-07:
To help users choose adequate hardware, we could populate [https://www.xcsoar.org/hardware/](XCSoar hardware page) with recommended and discommended phones.
@MaxKellermann do you have such a list? (even if it is just a draft)

MaxKellermann, 2023-08-07:
I can only say which Androids I use to develop/debug XCSoar, and that's: Google Pixel 6, Google Pixel 2, Cat S61.
After having many (non-XCSoar-related) frustrations with the many Androids I had from several vendors in the last 15 years, I'll never buy anything but a Google Pixel, ever again.

lordfolken, 2023-08-07:
Interestingly, and independently I arrived at the same conclusion. As evil as google is, they provide updates for 5 years and actually address issues. Plus they have an openable bootloader so you can run things like lineage, graphene or calyx. 

Pretty much all the rest of the androids, either provide no after sales or just one or two updates, contain spy ware or are modded so heavily that they become unusable. 

If you want such a list, please invent it. Make a user submit-table database and grade things such as compatibility and screen readability. But i really  think that this should be done by users, not the ones that already invest insane hours in xcsoar.

mlep, 2023-08-10:
I am unable to create a commit to https://github.com/XCSoar/website.
I do not understand what I am doing wrong.
Help, please.

MaxKellermann, 2023-08-10:
Fork, clone, commit to local clone, push to your fork, submit PR to official repository. As always with git.

mlep, 2023-08-10:
Thank you @MaxKellermann for your prompt reply.
I guess you are using the git command-line, while I was trying to do it on the Github website. It seems the way to achieve it are different. But well, I think I found where to click, hopefully.

lordfolken, 2023-09-22:
Documented as a known defect in certain phones.

closed 2023-09-22 by lordfolken