XCSoar/XCSoar #1280: Multiple Files for Maps / Waypoints / Airspaces

lordfolken, 2023-08-04:
XCSoar version
--------------
7.39 (Linux/Android/OpenVario)

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------
Currently its only possible to select 2 airspace and 2 waypoint files in the Site Configuration Dialog.

This is limiting, as I would like to configure more airspace and waypoint files. 
My flights often are through 3 countries, and that already requires reconfiguration in flight with the basic airspace and flight information.
Example: https://www.weglide.org/flight/281493

NOTAMS are also increasingly available via openair files, but these would count as additional files in the dialog, currently not configurable in every case. As an example here the swiss edabs (electric daily airspace bulletin)
https://www.segelflug.ch/safety-und-luftraum/luftraum-ii/luftraumdateien/

In general the airspace complexity has increased a lot in recent years, and both interaction with ATC and FlightPlan filing has become the norm. We need to increasingly behave like "normal" aircraft, meaning that we need a lot more information on the map.

(i have several efforts going on in the data backend, such as obstacle databases and openflightmap support, additional notam sources, airport circuits)

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
There was a PR once that seems to have been abandoned since: https://github.com/XCSoar/XCSoar/pull/471
I tried to revive it, but its too complex for me. 

What I think would be needed:
 * A multi select dialog (i.e. List but with a selectedable checkmarks infront of the filename)
 * Rewrite in the backend to deal with N files instead of one main and an additional file.

fraktal-sb, 2023-08-05:
Why not load the data from openaip homepage and set a filter to multiple countries? The download then contains the desired result in one file

lordfolken, 2023-08-05:
@fraktal-sb Why would you want to open a website, when all the openaip files (and others non-opanaip) are available in the backend? Of course _I_ can handmanage things, but that isn't the case for the majority of 100k users.

fraktal-sb, 2023-08-06:
it's just a workaround for people flying in the corner of 3 countries like me with Germany, France and Luxembourg. Not possible with currently 2 fields for input

remcoderksen, 2023-08-16:
Since I was looking through the beginners friendly threads I stumbled upon these ones.
#471 looks like it has been abandoned and closed recently. But before #471 there was #425. 
That one asked for extra fields on the Site files settings page. A simple solution I must say.
I had an extra working airspace file on the Site files settings page within a few minutes. So indeed 'beginners friendly' ;-)
But I read also someone asking for 7 extra airspace files. So where does it stop... Are people happy with 1,2 or 3 extra, or should there be 7 extra or is there someone else asking for 10 next week?! 
If this easy solution of adding extra fields is the way to go, then I think it would be wise to first agree on how many extra fields.

(I just added 2 at the end for testing purpose... clever naming can be thought about later)
[image]

(great result, all three airspaces in use)
[image]

llauner, 2024-01-01:
The folder where the files are stored should also be changed / reverted to a Public writable folder so that an external application can automatically download and update these files.
This is no longer possible when files are written to: Android/data/org.xcsoar/files/
See #1318

lordfolken, 2025-02-09:
> The folder where the files are stored should also be changed / reverted to a Public writable folder so that an external application can automatically download and update these files. This is no [...]

This is now possible if you want to do it, as the folder is now Android/media/org.xcsoar/

closed 2025-11-07 by lordfolken