XCSoar/XCSoar #1282: OGN Live Tracking

cjwoolley, 2023-08-08:
XCSoar version
--------------

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------

MaxKellermann, 2023-08-08:
Feature request is empty??

lyusupov, 2023-08-09:
@cjwoolley what is wrong with [**OGN Viewer - FLARM Radar**](https://play.google.com/store/apps/details?id=com.meisterschueler.ognviewer) app ?

https://github.com/Meisterschueler/ogn-viewer-android

> Android app that visualizes the aircraft traffic from [The Open Glider Network project (OGN)](http://glidernet.org/) on a map.
> OGN can receive signals from gliders, tow planes, helicopters, etc.
> This app is a flight radar for small aircraft with FLARM equipment.
> The decoded signals can be used with apps like **XCSoar on TCP port 4353.**

MaxKellermann, 2023-08-09:
What exactly does "OGN live tracking option" mean?

Your feature description confuses me even more. What does "achieve a single OGN trace" mean? What is an "ICAO address"? How does this feature help "ensure a continual live OGN flight trace"? How does it "improve the safety aspect"?

MaxKellermann, 2023-08-09:
Also missing in this feature request: the technical aspect. How is this supposed to work technically? How should XCSoar technically implement this "live tracking option"? What interefaces are there?

cjwoolley, 2023-08-10:
> @cjwoolley what is wrong with [**OGN Viewer - FLARM Radar**](https://play.google.com/store/apps/details?id=com.meisterschueler.ognviewer) app ?
>
> https://github.com/Meisterschueler/ogn-viewer-android

From my read, the OGN Viewer only displays OGN traffic and doesn’t integrate with XCSoar.
However, as you point out, XCSoar can read and display the OGN traffic via TCP port 4353.

My proposal is that when the flarm is out of range of an OGN ground station, XCSoar can communicate its position to the OGN server via the mobile cellular network. It’s essentially adding another communication path to the OGN server. Another way to think about it is that the mobile phone towers are doubling up as OGN ground stations.

cjwoolley, 2023-08-10:
> What exactly does "OGN live tracking option" mean?
> 
> Your feature description confuses me even more. What does "achieve a single OGN trace" mean? What is an "ICAO address"? How does this feature help "ensure a continual live OGN flight trace"? How does it "improve the safety aspect"?

“OGN live tracking option” means having the option to continually communicate your position with the OGN server as required, say on a 4 second interval, via the mobile phone cellular network to ensure a complete trace is available whether your flarm can access an OGN ground station or not.

To "achieve a single OGN trace" means switching between the OGN ground station supplying position data from the flarm to the OGN server and XCSoar supplying position data directly to the OGN server when the OGN ground station data is not available from the flarm. If neither the OGN ground station or the mobile phone cellular network is available then XCSoar should store the position data until it is able to forward it to the OGN server via the cellular network. See attached screen shots “Receiver:” switching between IGCDroid (mobile phone cellular network) and YKRY (OGN ground station). An ICAO hexadecimal address format is the common link between the two OGN sources (ground station & cellular network) and the OGN server.

[image]

“An ICAO address is an internationally recognised standard digital signature allocated to an aircraft or registration. https://en.wikipedia.org/wiki/Aviation_transponder_interrogation_modes
In the flarm’s case it’s known as the Radio-ID, OGN refers to it as the Device ID (https://ddb.glidernet.org/) and its common form here is a 6 digit hexadecimal number.

How does this feature help "ensure a continual live OGN flight trace"? – See above To “achieve a single OGN trace”

How does it "improve the safety aspect"? OGN is arguably the most used tracking data source with numerous live tracking apps using its data and thus out landing coordinates should be available if needed.

FYI: IGCDroid is a live tracking Android App which uses the mobile phone’s internal GPS for tracking. Integrating OGN live tracking into XCSoar makes the android device more efficient battery and charging wise. It also gives the XCSoar user the option to extend their OGN tracking beyond OGN ground station availability.

cjwoolley, 2023-08-10:
> Also missing in this feature request: the technical aspect. How is this supposed to work technically? How should XCSoar technically implement this "live tracking option"? What interefaces are there?

I can only supply the OGN recommended reference page for developers http://wiki.glidernet.org/dev/ to answer your questions. Alternatively, they can be contacted via email at [contact@glidernet.org](mailto:contact@glidernet.org)
The OGN’s data usage rules can be found here https://www.glidernet.org/ogn-data-usage/

MaxKellermann, 2023-08-10:
> I can only supply the OGN recommended reference page for developers http://wiki.glidernet.org/dev/ to answer your questions. 

This page doesn't answer any questions. It doesn't explain anything, and I had a quick look at the "C++ API" - and it's a big pile of garbage code which we certainly will not use.

MaxKellermann, 2023-08-10:
@cjwoolley you have a very complicated and verbose way of saying "I want XCSoar to periodically send its own position to an OGN server". Your big wall of text basically says just that and no more. You add lots of weasel words like "as required" but nothing is really required, and your verbose way of expressing yourself is just adding more confusion.

Now that we've arrived on understanding what you really want, what is still missing is the technical aspect. I don't see any protocol documentation anywhere and no API endpoint address. There's nothing on the Glidernet "dev" page; their C++ API is unusable garbage; and I don't see anything else anywhere.

But since XCSoar already has SkyLines live tracking, I'm not sure if this feature is worth my time. I hope there are other people who write that feature and submit it to me. (But PLEASE for the love of XCSoar DO NOT USE that official C++ API.)

lordfolken, 2023-08-10:
IMO an OGN client is overdue for xcsoar. 

I wasn't aware that you could send your own position directly to the ogn aprs server. I was under the impression that this is only possible with a receiver and its binary.

Seeyou has a nice description of the feature: 

https://naviter.com/2023/03/hybrid-live-tracking/

What i find especially nice:
"With a Wifi and LTE connection, an Oudie N or your phone running SeeYou Navigator becomes a hybrid, portable OGN transmitter. In places without OGN ground station network coverage, your phone becomes a “portable ground station,” relaying traffic information from Flarm and Fanet directly to OGN servers. This increases your situational awareness and the safety of your flying buddies to a whole new level."

The best documentation of how to send/recieve aprs packages to ogn:
http://wiki.glidernet.org/aprs-interaction-examples

lyusupov, 2023-08-11:
According to this: https://github.com/glidernet/ogn-aprs-protocol/blob/master/naviter_inputs.txt
- Naviter devices do NOT feed it's own traffic data directly to OGN servers.
Instead, they feed the data to Naviter's own service which further may (or may not) transfer it to the OGN system.

The XCSoar can supply own aircraft reports to SkyLines since long time ago - why don't the SkyLine's servers forward the data coming from XCSoar source down to the OGN just like the Naviter does for the data that comes from SeeYou Mobile app source ?

I always thought that XCSoar and SkyLines are affiliated projects which support each other, don't they ?

MaxKellermann, 2023-08-11:
> The best documentation of how to send/recieve aprs packages to ogn: http://wiki.glidernet.org/aprs-interaction-examples

wtf: "ask us for the algorithm."

> why don't the SkyLine's servers forward the data coming from XCSoar source down to the OGN

What would be the point of that? If XCSoar is already supposed to have an OGN connection, why should SkyLines have another one? (Another one for each and every XCSoar user, because each connection must have its own login)

lyusupov, 2023-08-11:
I assume that :
1) XCSoar has integrated 'network client' component for SlyLines service and
2) XCsoar does not have  integrated  'network client' component for OGN service yet

is that correct ?

If it is true - it could make sense to add a checkmark is the SkyLines dialog menu to let an XCSoar user make a decision  to give or not give a permission for SkyLines to share this particular traffic data beyond SkyLine's own ecosystem. When the checkmark is ON - the XCSoar device will have a chance to become visible on a map of other traffic info providers, such as OGN ecosystem.

lyusupov, 2023-08-11:
>  "ask us for the algorithm."

They probably address these words for some regular consumers with limited skills level...

There is actually no any secret in the 'APRS-IS passcode' algorithm that is in use by the OGN _(which is heavily based on 'Ham Radio' technologies, such as [APRS](https://en.wikipedia.org/wiki/Automatic_Packet_Reporting_System))_.

One can use a search engine request https://www.google.com/search?q=aprs+is+passcode

The PHP example is: https://github.com/magicbug/PHP-APRS-Passcode

C++ example is: https://github.com/gereic/GXAirCom/blob/master/lib/OGN/Ogn.cpp#L61

Online passcode generator: https://apps.magicbug.co.uk/passcode/

lordfolken, 2026-09-29:
Still open on purpose, and parked as backlog — not closing this.

If an OGN uplink (own position out, separate from XCSoar Cloud’s OGN downlink) is still something you want, the missing piece is scope rather than a flight repro: which APRS-IS messages, ownship only or traffic as well, and how it should sit next to SkyLines and LiveTrack24. A short note on that is enough to pick this up again.

dspreitz, 2026-10-07:
@lordfolken here is a scope proposal, based on how Naviter feeds OGN, and on what the Cloud receives today.

**What gets sent:** ownship only, and only when the FLARM ID is read from the connected FLARM. No relaying of FLARM/FANET traffic received by the phone. Relaying raises privacy questions and is not needed for the use case in this issue.

**Path:** via the XCSoar Cloud server, not directly from the app. The Cloud server already holds an APRS-IS connection to `aprs.glidernet.org` (today read-only with `N0CALL`/`-1`, `src/Cloud/Main.cpp`). With a callsign and its passcode it can also send. That means one connection instead of one per pilot, and the format can be fixed on the server without an app release. Naviter does the same: every OGNAVI message comes from their servers (`qAS,NAVITER3` in the live feed), not from the devices.

**Message format:** standard OGN-flavoured APRS aircraft beacon, following [Naviter_APRS_format.md](https://github.com/glidernet/ogn-aprs-protocol/blob/master/Naviter_APRS_format.md), with a new TOCALL `OGNXCS`:
```
FLR<id>>OGNXCS,qAS,<server>:/<hhmmss>h<lat>/<lon>'<track>/<speed>/A=<alt ft> !W<xy>! id<flags+type+id> +<climb>fpm <turn>rot
```
The source is the aircraft's own FLARM radio ID: `FLR<id>`, or `ICA<id>` if the FLARM transmits its ICAO address. OGN then merges it with the ground-station track of the same aircraft, which gives the single trace asked for here.

**Where the FLARM ID comes from (first step):** only from the connected FLARM. The FLARM driver already reads `PFLAC,R,RADIOID` on connect (`Device/Driver/FLARM/Mode.cpp`). The answer includes the ID type (ICAO or FLARM), which today is not kept in `FlarmHardware::radio_id` and would have to be. If XCSoar has no `RADIOID`, nothing is sent. That avoids duplicate tracks, IDs typed in wrong, and an ID scheme of our own. It does mean that a FLARM behind a vario (e.g. LXNAV passthrough) is not covered at first. That could later be added with a FLARM ID field in the plane profile.

**What the Cloud receives today, and what is missing.** It is not enough to forward on the server. The app sends a `FixPacket` with time, position, track, ground speed, airspeed, altitude and vario. The server keeps only position, altitude and track (`Tracking/SkyLines/Server.cpp`, `CloudServer::OnFix`):

| APRS field | Today | Needed |
|---|---|---|
| Source `FLR<id>` / `ICA<id>` | FLARM ID is not in the protocol. The app uses it only to filter itself from the traffic. | Send ID and ID type, e.g. in `FixPacket::reserved` with a new flag. Old servers ignore it. |
| Time | Sent; ignored by the server (`// TODO`) | Use it |
| Position, track | ✓ | – |
| Ground speed, climb rate | Sent; dropped by the server | Pass them through to the handler |
| Altitude | Baro altitude (QNH) preferred, GPS only as fallback | GPS altitude, like FLARM via OGN. Otherwise the merged track jumps between the two. |
| No-track / stealth / aircraft type | – | Opt-in flag from the app; the server also checks the OGN DDB flags |
| Turn rate | – | `0.0rot` |

**Rate:** today the app sends a fix to the Cloud only once per minute while flying (`CLOUD_INTERVAL`, `Tracking/SkyLines/Glue.cpp`). That is enough to find an outlanding, but gives a coarse trace next to the ground stations. With OGN opt-in, I propose one fix every 10 s. The cost stays small:
- **App:** a `FixPacket` is 48 bytes, 76 bytes with UDP/IP headers. That is about 27 KB of mobile data per flight hour.
- **Cloud server:** 1,000 pilots flying at the same time means 100 packets/s in, and about 12 KB/s of APRS out (about 120 bytes per beacon).

10 s is slower than FLARM via a ground station, but fills the gaps between stations well enough for a continuous track.

**Opt-in:** off by default, a separate switch next to the existing Cloud settings (see also #2691). The app sends the opt-in with every fix, so the server never forwards without it.

**SkyLines / LiveTrack24:** unchanged. OGN once assigned `OGSKYL` to SkyLines/xcsoar.org (with an API description from 2017), but I see no OGSKYL traffic in the feed today and `api.skylines.aero` does not answer. So I'd register `OGNXCS` with a format description in `glidernet/ogn-aprs-protocol` rather than revive `OGSKYL`.

**Steps if this scope is OK:**
1. agree with OGN (TOCALL PR with the format description);
2. app: keep the RADIOID type, opt-in switch, one fix every 10 s with opt-in;
3. Cloud protocol: FLARM ID and type, opt-in, GPS altitude;
4. Cloud server: keep time, speed and climb rate, check the DDB, send over APRS-IS.

dspreitz, 2026-10-07:
@meisterschueler Your view on this?
@snip @acasadoalonso Would OGNXCS as TOCALL and the NAVITER format work for you?

acasadoalonso, 2026-10-08:
Yes ... prepare a PR on the github repo

dspreitz, 2026-10-10:
@acasadoalonso the PR is up: glidernet/ogn-aprs-protocol#48 (TOCALL `OGNXCS`, APRS-IS login `XCSOAR`).

Three details changed from the scope above while writing the format down:
- **Sent with `qOR`**, as asked on glidernet/ogn-aprs-protocol#45: the Cloud server sends with no login in the packet, and consumers see `qAR,XCSOAR`.
- **No turn rate.** The fix does not carry one, and `0.0rot` would claim straight flight in a thermal.
- **Climb rate from the GPS altitude.** Today the app sends the TE vario, or the netto vario if there is no TE vario. For OGN, the `FixPacket` will need the GPS vertical speed and the GPS altitude.

The aircraft type comes from the FLARM's `ACFT` setting. An aircraft whose FLARM is set to `PRIV` or `NOTRACK` is not sent at all.

Once the format is accepted on the OGN side, steps 2–4 above follow in XCSoar.

acasadoalonso, 2026-10-11:
Merge done