XCSoar/XCSoar #1290: No flight log

csibeszke95, 2023-08-24:
XCSoar versions having and not having the problem
7.39
-------------------------------------------------

 don't know, but I have had logs until 29.06.2023

System information
------------------

Samsung A40 Android version 11
One UI version 3.1
Google play software refreshment 1.06.2023.

Steps to reproduce the behavior
-------------------------------

My daughter cancelled xc soar, I have downloaded again, however no flight log anymore from July. 

Expected behavior
-----------------

I just wanted to see my flight logs

Actual behavior
---------------

No flight log at all in the xcsoar data/logs

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

kobedegeest, 2023-08-25:
Might be related to #992

lordfolken, 2023-08-25:
If you dont select to keep the data when uninstalling.. then thats it. 
[image]

closed 2023-09-22 by lordfolken