XCSoar/XCSoar #1292: Map database causes freeze

Ozgrin, 2023-08-29:
XCSoar versions having and not having the problem
-------------------------------------------------

System information
------------------

Steps to reproduce the behavior
-------------------------------

Expected behavior
-----------------

Actual behavior
---------------

Do you have any idea what may have caused this?
-----------------------------------------------

Do you have an idea how to solve the issue?
-------------------------------------------

lordfolken, 2023-08-30:
I tested both israel maps and cannot reproduce a freeze.
Is it when you load any map? 
Can you try removing all files from Site config and try again?

lordfolken, 2023-08-31:
This happens in sim mode aswell, for both denmark and israel?
If you belive gps fix to be the problem, can you try it indoors in fly mode?
Can you try old releases? Htrps://download.
Xcsoar.org/releases

In any case can you try to get the log?
(Google files app
[image]
Then xcsoar.log

The maps are produced the same way, so there isnt really a difference between them.
The terrain loading process is asynchronus since a couple of versions.

MaxKellermann, 2023-08-31:
I think this could be the ANR bug that got reported on Google Play a few times. It's an endless loop in `Angle::AsBearing()` which I believe gets caused by an uninitialised Angle variable with a random value that is extremely big/small or even infinite or NaN. No matter how I tried, I was never able to find out where this uninitialised variable comes from. I always thought it could be some problem before the first GPS fix is received, and some part of XCSoar omits the check whether some field has a value.

Ozgrin, 2023-08-31:
I tried it in sim mode with the Denmark map - as soon as the map loaded the app fixed me on Denmark, started drawing the terrain and froze.

[xcsoar.log](https://github.com/XCSoar/XCSoar/files/12483795/xcsoar.log)

I'm attaching my log file.
I'll try to use an older release as well and report back.

Ozgrin, 2023-08-31:
Okay - 7.35 works! I guess I'll stick with this one for now.

MaxKellermann, 2023-08-31:
Does 7.36 work? For this piece of information to be useful for us, we need to know exactly what the first version with the problem was.

Ozgrin, 2023-08-31:
> Does 7.36 work? For this piece of information to be useful for us, we need to know exactly what the first version with the problem was.

7.36 works. 7.37 is the first one that doesn't (just tried both of them)

MaxKellermann, 2024-02-26:
I kept trying, and I also tried to inject bad values into XCSoar, but couldn't reproduce this problem. I guess I really need a device that is able to reproduce this.