XCSoar/XCSoar #1297: New Feature - Arrival Time Ring (ATR)

DavisChappins, 2023-09-05:
XCSoar version
--------------

Irrespective of version, but 7.x latest

What should XCSoar do differently, what functionality should be added?
----------------------------------------------------------------------

Add a new feature called Arrival Time Ring (ATR). 
  
  Similar to a fuel range ring found on powered aircraft.....
[image]

The Arrival Time Ring (ATR) would be a ring centered around your current goto waypoint, or task finish. The ATR ring would be specified for an arrival time (say 5pm). If you are inside the ring you would finish before 5pm and if you are outside of the ring you would finish after 5pm. The radius of the circle would decrease as you got closer to 5pm (in this example).

Radius of the circle (d) would be V_ete * T_remain
where V_ete is the input to the ETE info box and T_remain is the diff between now and the specified arrival time.

Radius would be a simple circle, nothing fancy. No value is gained by modulating the circle with wind or terrain. V_ete already probably accounts for wind.
Update rate can be lower as this does not need high fidelity.

The ring would have some text on its radius to inform the user what airport and what time it is indicating.
There would be a config setting to set the arrival time and also enable/disable.

What are you trying to do, what is the use case for the suggestion?
-------------------------------------------------------------------
If the forecast says the lift will die at 5pm and you want to be home by 5pm this is a graphical reminder that you need to be inside of this ring to finish or be home by 5pm. While flying an out/return task I constantly find myself calculating the return time, what time is it now, how much further I can go before I can turn back and still finish before the thermals quit. I would use this feature every flight. 
For task flying this would be a good graphical indication that you can make that last turnpoint before heading for home, or not, or how far to fly into the last AAT circle, and still make it home before thermals quit.

There would also be another ring for sunset.

DavisChappins, 2024-05-08:
6 months and not a single comment. oh well.

lordfolken, 2024-05-08:
I think the idea is good and makes sense. 
The easiest way to get it implemented, is to get started yourself.

DavisChappins, 2025-04-09:
PR for this feature. Unsure how the process goes since this is my first. It works great.
https://github.com/XCSoar/XCSoar/pull/1738

Implements a feature called Arrival Time Ring - a visual ring around the GOTO waypoint or last waypoint of the task. The ring radius is set by [ ( the difference between Arrival Time and local time ) * (the climb/cruise cross crountry speed used in the infobox FIN ETE) ].

Arrival time is set by a new menu item under Configuration > Map Display > Elements
The new field is called Arrival time ring and it is clickable to set a new time.

In use, the arrival time ring will be drawn on the map and will give a visual cue of your expected arrival time. Do you know the lift will die at 5pm (17:00). Set the arrival time to 1700 and if you are inside the ring, you'll make it home before 1700. If you are outside the ring you wont make it home by 1700. Again, radius and speeds are set by the same drivers of the FIN ETE infobox so the same assumptions to into the ring as that box (ie MC setting and all that).

DavisChappins, 2025-04-11:
https://github.com/XCSoar/XCSoar/pull/1738 has been updated to also include AAT time rings. So if your AAT time is 2:00 then a ring will slowly close in on your final waypoint if flying an AAT. The ring will be your 2 hour time so you can visually see how close or how far you are away from the 2:00 time.

DavisChappins, 2025-04-15:
Displayed is the arrival time ring (arrival 22:30) and the AAT time ring (AAT 1:15)

[image]

The arrival time ring indicates is described in a previous comment.
The AAT time ring shows the same information but set to the AAT time. So it gives you a nice visual indication of if you are going to be under time or over time.