XCSoar/XCSoar #1309: WIFI connection status no more visible on KOBO version 7.40

fraktal-sb, 2023-11-04:
XCSoar versions having and not having the problem
-------------------------------------------------

OK 7.38
NOK 7.40

System information
------------------

KOBO GLO

Steps to reproduce the behavior

see the following 2 screenshots from 7.38 and 7.40

-------------------------------

- Connect KOBO to local network
- Go to "Wifi" with the list of network devices
[image]

[image]

Expected behavior
-----------------

Connected network should be shown with status "connected" and network address

Actual behavior
---------------

Connection is established with "Wifi on" but the status is only "Saved and visible"

This is confusing compared to the previous version.

Connection is also shown with WPA 213 (7.40) and WPA 210 (7.38)

Bug ore feature?

Do you have any idea what may have caused this?
-----------------------------------------------

no

Do you have an idea how to solve the issue?
-------------------------------------------

no

ubx, 2023-11-06:
I confirm the problem on my Libra 2.

bisect ?

ubx, 2023-11-07:
Unfortunately, git bisect is difficult due to unavailable third-party libraries. A 'manual' bisect (installing different artifacts, like https://github.com/XCSoar/XCSoar/suites/16872196418/artifacts/963909733) shows that commit dee135e65cf948b549e288702505c8fb4a60523d is very likely the cause of the problem.

fraktal-sb, 2024-01-10:
OK in v7.41

closed 2024-01-10 by fraktal-sb