XCSoar/XCSoar #1312: NMEA messages send by XCSoar over BLE to FLARM are missing some parts ?

ludesert, 2023-11-07:
XCSoar versions having and not having the problem
-------------------------------------------------

Tried with version 7.40 only.

System information
------------------

Android 10

Steps to reproduce the behavior
-------------------------------

Connect to a BLE HM10 device that transmits data between FLARM and XSoar.
The data is transmitter without any issue from FLARM to XCSoar.

XCSoar seems to manage sending "simple" NMEA messages.
But some of them are not fully sent.

Here are some examples:

1) Task declaration fails at first message (sending the pilot name). Configured pilot name is "pilot name" (lowercase with space)
XCSoar sends the following sentence:
_$PFLAC,S,PILOT,piloame*20_
The correct sentence should be :
_$PFLAC,S,PILOT,pilot name*20_

The name has somehow changed. The result is:
_$ERROR,CKSUM*37_

Because the checksum "20" is not good for the sentence XCSoar sent. The pilot's name seems th have been changed after the calculation of the checksum (20 was good for the correct name).

2) I tried to configure the pilot name to "12345678901234567890".
Now XCSoar sends the sentence:
_$PFLAC,S,PILOT,12348901234567890*69_
The checksum should be 5D ...
Somehow, the pilot name has not been cut.

The result is obviously a checksum error message:
_$ERROR,CKSUM*37_

3) Same; with pilot name "1234567890 1234567890" (with a space)
XCSoar sends the sentence:
_$PFLAC,S,PILOT,1234890 1234567890*49_
The first part of the name misses the numbers 567...
Again a checksum error.

4) I noticed another problem like this while using the FLARM configuration menu (in Config/device/manage/configuration). I edited the range to max range (25500m).
XCSoar should send the sentence _$PFLAC,S,RANGE,25500*4A_.
But instead, it sends _$PFLAC,S,RANGE,2550A_.
Notice the absence of the * and the wrong checksum.

The result is _$PFLAC,A,RANGE,2500*6D_
Looks like the FLARM is not checking the checksum and accepting the configuration for 2500m. This is the value that appears when I reload the config settings.

This happens for every different value above 10000m.

I have no idea what's going on.

MaxKellermann, 2023-11-08:
Look at class HM10WriteBuffer. This class manages the buffer for writing to a HM10Port. The default MTU is 20 bytes, but can be changed if the device announces a different MTU. Your sentences are missing bytes after 19 bytes, which may be near a MTU boundary. The bug may be there somewhere.

ludesert, 2023-11-08:
I think you're right : the problem comes from the buffer size in XCSoar.
I tried LK8000 declaration (and FLARM configuration etc.) and I don't have this problem.

I tried to look at the class HM10WriteBuffer in XCSoar and compare it to LK8000's and it seems they use a variable named "maxChunkSize" ([see here](https://github.com/LK8000/LK8000/blob/master/android/src/org/LK8000/BluetoothGattClientPort.java#L81)) which is the equivalent of the MTU. It's also set to 20 bytes (see [here](https://github.com/LK8000/LK8000/blob/master/android/src/org/LK8000/HM10WriteBuffer.java#L118)) but for some reasons, it might be a problem of a number of bytes <20 of =< 20 ?

I'm sorry I will not be able to try and offer a correction as my programming knowledge is very limited and I don't know how to setup XCSoar's compilation environment on my computer/

ludesert, 2023-11-30:
Any idea ...?

closed 2023-12-21 by MaxKellermann